betaflight/betaflight #1448: Custom mixer not giving PWM output

atomiclama, 2016-11-03:
Needed to move one of my motors to a different pwm output so as usual did the changes in the mixer.
Now I get no output from the PWM to the ESCs, well the ESC no longer initialise.

mixer custom
mmix load quadx
save

no longer works

mikeller, 2016-11-03:
Version information would be helpful.

atomiclama, 2016-11-03:
Yeah sorry just thought about that.
It was CC3D built from master 2nd November.

I'm going to go back through some versions and see where it came in. 

Maybe something to do with the new IO mapping?

mikeller, 2016-11-03:
Well possible that something with custom mixer loading doesn't yet quite line up with resource mapping. But the good news is that you shouldn't have to do the custom mixer song and dance to reorder your motors any more. Just use `resource motor X PIN` twice to swap the pins for your motors...

atomiclama, 2016-11-03:
Thanks that was what I was going to look into to fix my problem, dead pin on uP. 
But then spotted this problem. I'll have a look until someone else beats me to it.

atomiclama, 2016-11-03:
Do you have a link to the nightly builds please, been a long time since I used stock FW.

mikeller, 2016-11-03:
http://andwho.sytes.net:8080/job/BorisB_BetaFlight/lastSuccessfulBuild/artifact/obj/

brycedjohnson, 2016-11-10:
Custom mixer also did not work for me to swap around pins.  Using build 682 - http://andwho.sytes.net:8080/job/BorisB_BetaFlight/682/

Going to try the resource motor x pin next

brycedjohnson, 2016-11-10:
Swapping motors with the resource motor x pin works pretty slick!

PR #1584 (2016-11-19): Fixed a problem when the motor count for motorInit came from custom mixer 
fixes #1448 
Not flight tested but PWM outputs checked with Logic analyser, oneshot, normal pwm, both with mixer quadx and custom.

1. took the motor count from the loading of the custom mixer table.
2. Moved the mixerconfig to before motorInit as motorInit needs the
motorCount.
3. removed pwmIsSynced checks as this was a order of init problem, and
not required as pwmCompleteMotorUpdate protects itself.
closed 2016-11-19 by blckmn