betaflight/betaflight #1449: Fix issue with -flto optimiser

borisbstyle, 2016-11-03:
Currently this optimiser flag is disabled due to some unexplained issues in DMA function.

See more info in this PR
https://github.com/betaflight/betaflight/pull/1444

ledvinap, 2016-11-03:
for reference,, copy from https://github.com/betaflight/betaflight/pull/1444#issuecomment-258076388 by @borisbstyle 

>The issue in here is that when you reserve some memory for DMA and you let DMA push it to periphirals that it will keep repeating the last part for some period. This happens very consistantly on the same GPIOs unregarding of how and when you perform that part of the code. This occurs outside the CPU activity and I really dont see the reason why it would do that. I rather think the optimiser is shifting / optimising parts of the memory what are crucial for DMA than its just an ordinary memory corruption.

borisbstyle, 2016-11-03:
@ledvinap 
issue fixed! 
https://github.com/betaflight/betaflight/commit/7100454a1d26582e9830056600c3d4b015c36b4c

digitalentity, 2016-11-06:
@borisbstyle interesting, why disabling LTO fixed this.

closed 2016-11-06 by digitalentity