betaflight/betaflight #1450: Motor output PWM incorrect.

atomiclama, 2016-11-03:
Testing local build from this master this morning. I can see that the pwm out on the CC3D is broken, no other targets to test on.

As you can see from the attached pic motors 0-2 have not been adjusted to the oneshot scaling and are outputting a 1ms pulse.

Motor 3 is correct.  
[image]

I have had a quick look through the resource IO mapping and can find nothing obvious.

borisbstyle, 2016-11-03:
looks like inverted for oneshot on 1-3

atomiclama, 2016-11-03:
Nah it's the 1ms default pulse width.

atomiclama, 2016-11-03:
Ok it's a conflict with TIM4 used for PWM input.
If I disable PWM rx and use serial rx then they are ok.

borisbstyle, 2016-11-03:
@atomiclama Yeah thought something like that. But I cant remember that changed any recently?

mikeller, 2016-11-03:
Could it be that any of the resource reshuffling done for DShot is causing this? Or another undesired side effect of disabling link time optimisation?

blckmn, 2016-11-03:
It is most likely the code to prevent timer conflicts

I only tested it on PPM, will have a quick look and see if it is working for PWM

blckmn, 2016-11-03:
Actually ... there is no code to prevent timer clash for PWM. if you use PWM you only have standard (1ms) output allowed...

blckmn, 2016-11-03:
I am thinking we disable PB6 for input as a default on CC3d - it can be remapped if needed.

atomiclama, 2016-11-04:
Ok so this is a known limitation of the CC3D and similar.

Would it be worth putting a block on setting oneshot if PWM is set. 
Or a change in default RX method as oneshot is enabled as default.
Or just a note in the wiki.

borisbstyle, 2016-11-04:
No that can probably be resolved by rearranging the timers. Will have a
quick look and let you test

Saying that I think nobody should be using PWM nowadays ;p

atomiclama, 2016-11-04:
Agreed but some people still are.

I only spotted this as I was testing out the resource mapping on a default build on the bench. Would have missed it if I had gone straight to my quad with iBus

Yeah sure I'll test it.

borisbstyle, 2016-11-04:
@atomiclama Thats indeed a danger you get now. Nobody is even testing PWM on all targets. Even I dont have PWM setup to test. I expect many issues in there on various targets. Its just waste of resource supporting PWM.

borisbstyle, 2016-11-04:
btw I agree configurator would disallow PWM on targets with expected issues

borisbstyle, 2016-11-04:
also try unsynced pwm! it might work!

mikeller, 2016-11-06:
> btw I agree configurator would disallow PWM on targets with expected issues

The problem is we are talking 20+ features times 20+ boards times a dozen or so firmware versions, which makes for a staggering number of feature combinations that the configurator would have to know about. My hope would be that the to be decided upon 'return supported hardware features' command on the FC would return 'supported / unsupported' and 'permanent / dynamic', forcing the configurator to reevaluate availability of features after other features have changed.
This wold also tie in nicely with dynamic resource allocation (i.e. no point in offering PPM as an option of no pin has been assigned to PPM).

Noctaro, 2016-11-06:
> Saying that I think nobody should be using PWM nowadays ;p
hmmmm.....agreed, but lots of the available china multicopter come with PWM receiver and PWM esc on a CC3D setup... I wonder if this could have something to do with the "rolls of death" thats occuring at full stick deflection rolls on my CC3D with PWM transmitter and PWM ESC? (running unsynced at 400hz)

closed 2016-12-01 by blckmn