betaflight/betaflight #1461: Baro on NAZE32 REV6 stops in 3.0.1

nigelsim, 2016-11-05:
Using a NAZE32 REV6 the barometer works in 3.0.0 but doesn't in 3.0.1.

Miniwen, 2016-11-05:
Actually for Naze32 rev6 barometer and magnetometer are disabled since version 3.0 of Betaflight (flashinf 2.9.0.1 brings them back as well as CLI comands )changing sliders under configuration tab doesn't make any difference. CLI comands not recognized.

mikeller, 2016-11-05:
Yes. Support for barometer and magnetometer was removed from NAZE in 3.0 since this target was running out of flash space, and new features like the new PID controller and filters needed additional space.

Miniwen, 2016-11-05:
And which FC's still support Baro and Mag? SPF3 does?

mikeller, 2016-11-05:
The small flash size problem and removal of features so far only occurs on flight controllers with an F1 CPU, the F3 and F4 CPUs that are used have more flash.

blckmn, 2016-11-09:
It may be possible to add baro back in future if there's  some other lesser used functions that are more sacrificial to free up space.

closed 2016-11-09 by blckmn

mikeller, 2016-11-09:
Not sure how much value is in that though. All the (acro-) NAZEs that I've used, the baro readings were always going all over the show, making the data all but useless.

nigelsim, 2016-11-09:
I guess the best outcome for this ticket would be that this been noted in the NAZE notes page. The only time I ever actually used the baro on 3.0.0 was to help get a sense of the height using the telemetry. I never actually did baro mode.

nipunnawanjana1, 2018-02-20:
it work for me https://youtu.be/9Es1NzxLyWE