betaflight/betaflight #1463: BrainFPV RE1 flipping over

oliv98, 2016-11-05:
Hi folks, having some strange issue with RE1, DYS20A, BF 3.0.1,Tmotor F40, 4S

just finished mounting the quad and I had some strange behavior in angle mode. Did it twice. 
The quad is just flipping over without me doing anything significant stick input while hovering since a minute.
I've included blackbox log, this is at the end of the last log(the 10th). Any help appreciated.

[blackbox_log_2016-11-05_160640.TXT](https://github.com/betaflight/betaflight/files/573074/blackbox_log_2016-11-05_160640.TXT)

brat002, 2016-11-13:
Try to lower your pids. I had same issue with sparky 2 board.

borisbstyle, 2016-12-20:
I will close this one as other users report correct behaviour of this board.

Checklist:
Check motor positions if those are correct. Motor1 = also motor 1 etc
Check motor spin directions
Check prop direction

closed 2016-12-20 by borisbstyle