betaflight/betaflight #1470: MSP_CANVAS or else

jflyper, 2016-11-06:
This is for discussion of whether to allocate a new MSP message type MSP_CANVAS (this particular name is a subject to change, perhaps MSP_DISPLAYPORT) to support display port abstraction on OSDs on serials.

Before CMS and associated drivers (#1422) get finalized, I would like to hear community's comments/opinions on whether to allocate a new MSP message for sending CMS text drawing actions to external OSDs.

Points of considerations:

- Consumption of scarce resource (1-byte MSP message type space)
- Existence of similar message types, e.g., MSP_OSD
- Clarity and extensibility of the display port abstraction over MSP
- ...

Any thoughts?

mikeller, 2016-11-06:
I think it's a good idea. `MSP_OSD` cannot easily be repurposed, since it's being used by legacy OSD devices, and will probably be so for a while.

We could probably avoid using a new MSP code by reallocating `MSP_SPECIAL_PARAMETERS = 98`, which was only used in a hack to support the by now retracted (due to a dangerous bug) release 2.9.0.

martinbudden, 2016-11-06:
Althought the MSP message types are scarce, they are there to be used and this is certainly a worthwhile usage of them. Repurposing an existing type, as @mikeller points out, is problematic.

So I think we should allocate a new MSP message type.

jflyper, 2016-11-06:
Okay, what about the name? MSP_CANVAS is already a history now :) MSP_DISPLAYPORT?

llambkin, 2016-11-07:
Hahaha, I just cloned a build tonight. It seems that MSP_DISPLAYPORT wins :D

jflyper, 2016-11-09:
Yeap :)

closed 2016-11-09 by jflyper