betaflight/betaflight #1471: sbus not working in 3.0.1 on Flip32 Omnibus (F3)

ChrisWithem, 2016-11-06:
I recently received a Flip32 Omnibus FC, promptly updating the f/w to most recent (3.0.1) but was unable to get sbus to work.  I'm using a Futaba T10CHP tx with FrSky TFR4SB rx.  After toggling ON "Serial RX" on UART3, and confirming Receiver Mode and Serial Receiver Provider were correct, I still failed to get any activity on the Receiver tab.  Even tried re-flashing and full-chip-erase.  I eventually flashed 3.0.0 back on the FC and sbus worked.

jflyper, 2016-11-06:
This looks like a support request. Can you close this and come the RCG thread? You can copy & paste the question.

[Omnibus F3AIO](https://www.rcgroups.com/forums/showthread.php?t=2711617&page=47#post36128027)

ChrisWithem, 2016-11-06:
Whoops.  Sorry about that.  Will do.

closed 2016-11-06 by ChrisWithem

Tomcatt, 2018-03-03:
I'm in the same boat as well.  I can bind it all day but there's a disconnect from the receiver to the board.  I've installed the latest version of the OMNI v3.3.0.