betaflight/betaflight #1473: latest commits bf 3.1.0 - make process shows errors and ledstrip does not work

TheAngularity, 2016-11-06:
checked out latest commits ... make target SPRACINGF3EVO brings following errors and the led strip is not working anymore 👎 

```
$ make TARGET=SPRACINGF3EVO
make -j ./obj/betaflight_3.1.0_SPRACINGF3EVO.hex
make[1]: Verzeichnis „/home/betaflight“ wird betreten
%% startup_stm32f30x_md_gcc.S
%% system_stm32f30x.c
.
.
.
%% main.c
%% encoding.c
./src/main/target/SPRACINGF3EVO/target.c:27:5: warning: missing initializer for field 'dmaChannel' of 'timerHardware_t {aka const struct timerHardware_s}' [-Wmissing-field-initializers]
     { TIM8,  IO_TAG(PA15), TIM_Channel_1, TIM8_CC_IRQn,         TIM_USE_PPM,   0, GPIO_AF_2 },  // PPM
     ^
In file included from ./src/main/target/SPRACINGF3EVO/target.c:23:0:
./src/main/drivers/timer.h:101:26: note: 'dmaChannel' declared here
     DMA_Channel_TypeDef *dmaChannel;
                          ^
./src/main/target/SPRACINGF3EVO/target.c:28:5: warning: missing initializer for field 'dmaChannel' of 'timerHardware_t {aka const struct timerHardware_s}' [-Wmissing-field-initializers]
     { TIM2,  IO_TAG(PA0),  TIM_Channel_1, TIM2_IRQn,            TIM_USE_MOTOR, 1, GPIO_AF_1 },  // PWM1
     ^
In file included from ./src/main/target/SPRACINGF3EVO/target.c:23:0:
./src/main/drivers/timer.h:101:26: note: 'dmaChannel' declared here
     DMA_Channel_TypeDef *dmaChannel;
                          ^
./src/main/target/SPRACINGF3EVO/target.c:29:5: warning: missing initializer for field 'dmaChannel' of 'timerHardware_t {aka const struct timerHardware_s}' [-Wmissing-field-initializers]
     { TIM2,  IO_TAG(PA1),  TIM_Channel_2, TIM2_IRQn,            TIM_USE_MOTOR, 1, GPIO_AF_1 },  // PWM2
     ^
In file included from ./src/main/target/SPRACINGF3EVO/target.c:23:0:
./src/main/drivers/timer.h:101:26: note: 'dmaChannel' declared here
     DMA_Channel_TypeDef *dmaChannel;
                          ^
./src/main/target/SPRACINGF3EVO/target.c:30:5: warning: missing initializer for field 'dmaChannel' of 'timerHardware_t {aka const struct timerHardware_s}' [-Wmissing-field-initializers]
     { TIM15, IO_TAG(PA2),  TIM_Channel_1, TIM1_BRK_TIM15_IRQn,  TIM_USE_MOTOR, 1, GPIO_AF_9 },  // PWM3
     ^
In file included from ./src/main/target/SPRACINGF3EVO/target.c:23:0:
./src/main/drivers/timer.h:101:26: note: 'dmaChannel' declared here
     DMA_Channel_TypeDef *dmaChannel;
                          ^
./src/main/target/SPRACINGF3EVO/target.c:31:5: warning: missing initializer for field 'dmaChannel' of 'timerHardware_t {aka const struct timerHardware_s}' [-Wmissing-field-initializers]
     { TIM15, IO_TAG(PA3),  TIM_Channel_2, TIM1_BRK_TIM15_IRQn,  TIM_USE_MOTOR, 1, GPIO_AF_9 },  // PWM4
     ^
In file included from ./src/main/target/SPRACINGF3EVO/target.c:23:0:
./src/main/drivers/timer.h:101:26: note: 'dmaChannel' declared here
     DMA_Channel_TypeDef *dmaChannel;
                          ^
./src/main/target/SPRACINGF3EVO/target.c:32:5: warning: missing initializer for field 'dmaChannel' of 'timerHardware_t {aka const struct timerHardware_s}' [-Wmissing-field-initializers]
     { TIM3,  IO_TAG(PA6),  TIM_Channel_1, TIM3_IRQn,            TIM_USE_MOTOR, 1, GPIO_AF_2 },  // PWM5
     ^
In file included from ./src/main/target/SPRACINGF3EVO/target.c:23:0:
./src/main/drivers/timer.h:101:26: note: 'dmaChannel' declared here
     DMA_Channel_TypeDef *dmaChannel;
                          ^
./src/main/target/SPRACINGF3EVO/target.c:33:5: warning: missing initializer for field 'dmaChannel' of 'timerHardware_t {aka const struct timerHardware_s}' [-Wmissing-field-initializers]
     { TIM3,  IO_TAG(PA7),  TIM_Channel_2, TIM3_IRQn,            TIM_USE_MOTOR, 1, GPIO_AF_2 },  // PWM6
     ^
In file included from ./src/main/target/SPRACINGF3EVO/target.c:23:0:
./src/main/drivers/timer.h:101:26: note: 'dmaChannel' declared here
     DMA_Channel_TypeDef *dmaChannel;
                          ^
./src/main/target/SPRACINGF3EVO/target.c:34:5: warning: missing initializer for field 'dmaChannel' of 'timerHardware_t {aka const struct timerHardware_s}' [-Wmissing-field-initializers]
     { TIM3,  IO_TAG(PB0),  TIM_Channel_3, TIM3_IRQn,            TIM_USE_MOTOR, 1, GPIO_AF_2 },  // PWM7
     ^
In file included from ./src/main/target/SPRACINGF3EVO/target.c:23:0:
./src/main/drivers/timer.h:101:26: note: 'dmaChannel' declared here
     DMA_Channel_TypeDef *dmaChannel;
                          ^
./src/main/target/SPRACINGF3EVO/target.c:35:5: warning: missing initializer for field 'dmaChannel' of 'timerHardware_t {aka const struct timerHardware_s}' [-Wmissing-field-initializers]
     { TIM3,  IO_TAG(PB1),  TIM_Channel_4, TIM3_IRQn,            TIM_USE_MOTOR, 1, GPIO_AF_2 },  // PWM8
     ^
In file included from ./src/main/target/SPRACINGF3EVO/target.c:23:0:
./src/main/drivers/timer.h:101:26: note: 'dmaChannel' declared here
     DMA_Channel_TypeDef *dmaChannel;
                          ^
./src/main/target/SPRACINGF3EVO/target.c:36:5: warning: missing initializer for field 'dmaChannel' of 'timerHardware_t {aka const struct timerHardware_s}' [-Wmissing-field-initializers]
     { TIM2,  IO_TAG(PB10), TIM_Channel_3, TIM2_IRQn,            TIM_USE_MOTOR, 1, GPIO_AF_1 }, // RC_CH4 - PB10 - *TIM2_CH3, UART3_TX (AF7)
     ^
In file included from ./src/main/target/SPRACINGF3EVO/target.c:23:0:
./src/main/drivers/timer.h:101:26: note: 'dmaChannel' declared here
     DMA_Channel_TypeDef *dmaChannel;
                          ^
./src/main/target/SPRACINGF3EVO/target.c:37:5: warning: missing initializer for field 'dmaChannel' of 'timerHardware_t {aka const struct timerHardware_s}' [-Wmissing-field-initializers]
     { TIM2,  IO_TAG(PB11), TIM_Channel_4, TIM2_IRQn,            TIM_USE_MOTOR, 1, GPIO_AF_1 }, // RC_CH3 - PB11 - *TIM2_CH4, UART3_RX (AF7)
     ^
In file included from ./src/main/target/SPRACINGF3EVO/target.c:23:0:
./src/main/drivers/timer.h:101:26: note: 'dmaChannel' declared here
     DMA_Channel_TypeDef *dmaChannel;
                          ^
./src/main/target/SPRACINGF3EVO/target.c:38:5: warning: missing initializer for field 'dmaChannel' of 'timerHardware_t {aka const struct timerHardware_s}' [-Wmissing-field-initializers]
     { TIM1,  IO_TAG(PA8),  TIM_Channel_1, TIM1_CC_IRQn,         TIM_USE_LED,   1, GPIO_AF_6 },  // GPIO_TIMER / LED_STRIP
     ^
In file included from ./src/main/target/SPRACINGF3EVO/target.c:23:0:
./src/main/drivers/timer.h:101:26: note: 'dmaChannel' declared here
     DMA_Channel_TypeDef *dmaChannel;
                          ^
%% filter.c
%% maths.c
.
.
.
%% usb_pwr.c
%% usb_io.c
LD betaflight_SPRACINGF3EVO.elf
arm-none-eabi-size ./obj/main/betaflight_SPRACINGF3EVO.elf
   text    data     bss     dec     hex filename
 156856    2052   27120  186028   2d6ac ./obj/main/betaflight_SPRACINGF3EVO.elf
arm-none-eabi-objcopy -O ihex --set-start 0x8000000 obj/main/betaflight_SPRACINGF3EVO.elf obj/betaflight_3.1.0_SPRACINGF3EVO.hex
make[1]: Verzeichnis „/home/betaflight“ wird verlassen

```

blckmn, 2016-11-06:
https://github.com/betaflight/betaflight/pull/1478 resolves this.

TheAngularity, 2016-11-06:
@blckmn 
not really ... errors are done but ledstrip not working :(

blckmn, 2016-11-07:
Can you output the results of the `resource` cli command?

If LED_STRIP is not listed, can you reset your settings.

TheAngularity, 2016-11-07:
here the output ... i think all is good

```
# resource
resource BEEPER C15
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 A06
resource MOTOR 6 A07
resource MOTOR 7 B00
resource MOTOR 8 B01
resource MOTOR 9 B10
resource MOTOR 10 B11
resource PPM A15
resource LED_STRIP A08
```

TheAngularity, 2016-11-08:
@blckmn 
output from your latest pr #1490 ... ledstrip still does not work

```
Entering CLI Mode, type 'exit' to return, or 'help'

# resource list
Currently active IO resource assignments:
(reboot to update)
----------------------
A00: MOTOR1 OUT
A01: MOTOR2 OUT
A02: MOTOR3 OUT
A03: MOTOR4 OUT
A04: ADC BATTERY
A05: FREE 
A06: FREE 
A07: FREE 
A08: LED_STRIP OUT
A09: SERIAL1 UART TX
A10: SERIAL1 UART RX
A11: FREE 
A12: FREE 
A13: FREE 
A14: FREE 
A15: SERIAL2 UART RX
B00: FREE 
B01: FREE 
B02: FREE 
B03: SPI1 SCK
B04: SPI1 MISO
B05: SPI1 MOSI
B06: I2C1 SCL
B07: I2C1 SDA
B08: LED1 OUT
B09: MPU CS
B10: FREE 
B11: FREE 
B12: SDCARD CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C13: MPU EXTI
C14: SDCARD IN
C15: BEEPER OUT
F00: FREE 
F01: FREE 
F04: FREE 

Currently active DMA:
DMA1 Channel 1: FREE
DMA1 Channel 2: LED_STRIP
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC2
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

Use: 'resource' to see how to change resources.

# 
```

TheAngularity, 2016-11-08:
@blckmn 
searching on jenkins and found the the last build where the ledstrip still works
[betaflight_3.1.0-656-537e4ce](http://andwho.sytes.net:8080/job/BorisB_BetaFlight/656/)

TheAngularity, 2016-11-09:
merged latest commits ... now i'm on b2ad1a2 but ledstrip still does not work

any news @blckmn fixed this bug

TheAngularity, 2016-11-10:
nice 👍 ... @blckmn latest #1524 fixed the ledstrip bug on my evo ... thx

blckmn, 2016-11-10:
That's great news. Credit goes to @sambas for the discovery of the fix. Seems there's a slight difference in how the DMA controller functions between F1, F3 and F4. A few gotchas.

closed 2016-11-10 by blckmn