betaflight/betaflight #1475: Current master for 3.1.0 does not work with SERIALRX (and maybe more?)

geebles, 2016-11-06:
Hi, when using the latest 3.1.0 master branch (e.g. build here http://andwho.sytes.net:8080/job/AndWho_Betaflight/382/) with target SPRACINGF3, with SERIALRX on UART3, set to either SPEK1024 or SPEK2048, on the 'RX' tab when the radio is powered on ALL channels change to '988' (minimum value) and remain there until the radio is powered off (where they jump back to their failsafe positions).

Other targets/uarts/RX protocols may be affected, I have only tested the above hardware. Flashing to the latest 3.0.1 results in RX working perfectly (so no hardware issue here).

AndersHoglund, 2016-11-06:
Confirmed on REVO. Spektrum Sat works fine on 3.0.1. 
With 3.1 I it is broken. I get a value of 988 on all channels. Turning Tx off results in 1500 on R/P/Y, 885 on Throttle and 988 on all remaining AUX. So there is some connection, makes failsafe (?) kick in.

    # diff
    # version
    # Betaflight/REVO 3.1.0 Nov  6 2016 / 13:04:50 (578bf91)
    # map
    map TAER1234
    # serial
    serial 2 64 115200 57600 0 115200
    # master
    set serialrx_provider = SPEK2048

AndersHoglund, 2016-11-08:
Re-test on latest master HEAD ( 3fe8e4b) , Serial Sat Rx still very broken on REVO and SPRACINGF3.

skullq, 2016-11-08:
I also have same issue in 3.1.0 build#666 . Flip32 F4(REVO). Spektrum Sat on UART1 doesn't work and all values on receiver tabs got to 988. Going back to 3.0.1 with same setup no issue.

stupport81, 2016-11-09:
PPM RX input on SPRacingF3 target not working for me on 3.1.0 build 666. 3.0.1 is working fine.

AndersHoglund, 2016-11-09:
Tried serial rx on REVO again, same result. Started testing with a fresh flash and "Reset Settings", only added UART3 Serial Rx, Spektrum2048. Still fails.  
Then added a few things, DSHOT150, LED-strip. No change.  From what I can see there is no DMA conflict. Resource list output:

    Currently active DMA:
    DMA1 Stream 0: FREE
    DMA1 Stream 1: MOTOR3
    DMA1 Stream 2: MOTOR1
    DMA1 Stream 3: SERIAL3
    DMA1 Stream 4: LED_STRIP
    DMA1 Stream 5: FREE
    DMA1 Stream 6: MOTOR2
    DMA1 Stream 7: MOTOR
    DMA2 Stream 0: FREE
    DMA2 Stream 1: FREE
    DMA2 Stream 2: FREE
    DMA2 Stream 3: FREE
    DMA2 Stream 4: ADC1
    DMA2 Stream 5: FREE
    DMA2 Stream 6: FREE
    DMA2 Stream 7: SERIAL1
    
    Use: 'resource' to see how to change resources.
    
    # version
    # Betaflight/REVO 3.1.0 Nov  9 2016 / 21:38:48 (054b399)

martinbudden, 2016-11-09:
@AndersHoglund , I'm suspicious of this PR https://github.com/betaflight/betaflight/pull/1162 , in particular this line: https://github.com/betaflight/betaflight/pull/1162/files#diff-2da41031eb7d7f076567ebda42e5223eR151

Do you have an RSSI channel set? If not, could you try setting one and see if that fixes the proplem.

AndersHoglund, 2016-11-09:
Maestro! You have got something there. RSSI ch was "disabled".
RSSI ch 1 did not help. but RSSI ch 12 did. But it also needed a re-boot. Just changing channel does not help.

martinbudden, 2016-11-09:
@AndersHoglund , can you try this to see if it fixes the problem - https://github.com/betaflight/betaflight/pull/1513.

Hopefully it should work if RSSI ch is set or disabled.

AndersHoglund, 2016-11-09:
Yes, PR #1513 fixes this issue. 
Solution available on master from commit (458de15f910f234160c25a8c8c1334575f20839d) ,and
Jenkins build 682, http://andwho.sytes.net:8080/job/BorisB_BetaFlight/682/

martinbudden, 2016-11-09:
Closing, fixed by PR https://github.com/betaflight/betaflight/pull/1513

closed 2016-11-09 by martinbudden