betaflight/betaflight #1485: BF 3.1.0 Piko BLX (Kombini) - cant save Dhsot setting

Guidus93, 2016-11-07:
Hei, is there someone that have flashed bf 3.1 and tested dshot? I've flashed kombini with bf 3.1,aikon 0a with dshot 16.42 and using configurator 1.8.5, in the configurator I can see dshot protocol but cant save it,in the cli the dshot protocol are not allowed values in pwm motor protocol... I've read a lot on rc group and seems that problem is because the fc is not support by dshot for the moment, but I've also see that someone use dshot with kombini.... The motor spin with multishot protocol but havent fly yet, other stuff seems work good

borisbstyle, 2016-11-07:
When dshot is not supported on the board it will jump back. It is desired behaviour until dshot is supported

closed 2016-11-07 by borisbstyle

gotnull, 2016-12-19:
Is the Piko BLX board compatible with Dshot though and it's just a software limitation at this time?

mikeller, 2016-12-19:
> Is the Piko BLX board compatible with Dshot though and it's just a
> software limitation at this time?

The Piko BLX board does support DShot, but it is not straightforward to set
up. https://github.com/betaflight/betaflight/wiki/BetaFlight%20and%20Dshot
has instructions specific to the Piko BLX.