betaflight/betaflight #1508: No roll pid authority REVO betaflight 3.0.1

phredieFPV, 2016-11-09:
I can't work out what the cause of this is. Reflashing the board fixes it for a while but within the next 2-3 flights the issue will start again.

Stick commands work but the quad will not hold its attitude on the roll axis. Blackbox log shows that without stick inputs motors 1&3 and 2&4 are always the same value (see attached blackbox log and dump), The blackbox log is of me manually roll/pitch/yawing the quad without props on.

Any help would be appreciated. Thanks!
[blackbox_log_2016-11-05_111253.TXT](https://github.com/betaflight/betaflight/files/580925/blackbox_log_2016-11-05_111253.TXT)
[Dump.txt](https://github.com/betaflight/betaflight/files/580926/Dump.txt)

fftunes, 2016-12-09:
Hm i noticed something like that for short moments while flying but never caught it in a log. 
Felt like losing i-term or something, while it still reacted fine to sticks.
Building another rig right now, again with rtfq revo... i'll see.

martinbudden, 2016-12-10:
@borisbstyle , see the above comment. Could this be related to the ITerm rebase changes you made?

mikeller, 2016-12-10:
@martinbudden The original poster is on 3.0.1, so certainly not for him.

martinbudden, 2016-12-10:
@mikeller , yes, I was talking about the second poster, @fftunes

fftunes, 2016-12-12:
Sorry i meant 3.0.1 too. Haven't tried any newer yet on that specific quad.

borisbstyle, 2016-12-19:
you roll and even pitch axis doesnt do anything on the blackbox.
Looks like receiver/transmitter issue

closed 2016-12-19 by borisbstyle

RBeaubien, 2016-12-31:
I have the same issue on 3.0.1.  I get 1-3 flights then the motors no longer respond to pitch or roll changes (one or the other).  I had this happen to 3 REVO clones and I thought I had bad flight controllers.  I have since discovered doing a full erase and reflashing fixes the problem for a while.  I have 2 flight controllers in this state if you want me to test something.