betaflight/betaflight #1537: Multishot RGB LED Motor Clicking

spudw, 2016-11-11:
I recently swapped out my old ESCs for ones that support MultiShot and I noticed a strange behavior.  When MultiShot is selected and I turn on my RGB LEDs I get a clicking sound from my motors.  It only seems to happen with MultiShot which leads me to believe it is a CPU/loop/timing issue.  I am running BetaFlight 3.0.1 on a RMRC Dodo Rev 3b board with MRM 20a Zues (LittleBee) ESCs and using a 4khz Gyro loop and 2khz PID loop.  
Here is a video that illustrates the issue.  https://www.youtube.com/watch?v=FUCim_7H0ec

spudw, 2016-11-11:
Issue subsides, but not completely, when CPU load is reduced (running 2khz Gyro and 2khz PID).

borisbstyle, 2016-12-19:
Please try 3.1 development version. Can you also find out what motor is clicking?

Please open a new issue if it still persists.

closed 2016-12-19 by borisbstyle

bombarie, 2017-03-11:
@spudw I had the same and solved it by decoupling the PID loop and motor PWM speed. The clicking stopped when I switched on "Motor speed seperate from PID speed". I set it to 8000. After that, I could put my gyro update back to 8khz and I put my pid loop at 4khz for the time being. No clicking anymore. It indeed suggests a cpu load issue. Hope this helps.

my setup:
- Racerstar BR2205 2300KV motors
- Racerstar MS35A ESC's @ BLHELI_S 16.65
- SP Racing F3 EVO FC with SPI mpu9250 chip
- Betaflight 3.1.6