betaflight/betaflight #1538: Include docker-file to allow people to build in a consistent environment

andymac4182, 2016-11-12:
It would be great if there could be a docker-file included as part of betaflight that would give a standard environment to build the code with. Someone has perviously created one but it appears to be out of date. https://github.com/mikeller/docker-betaflight-build/blob/master/Dockerfile

andymac4182, 2016-11-12:
I have updated the docker file and confirmed it can build the current code. https://github.com/andymac4182/betaflight/blob/docker-build/Dockerfile 

This will ensure that no matter which operating system people are using as their main OS they can build the code in a reliable way without installing anything other than docker.

mikeller, 2016-11-12:
Can you please submit a pull request against https://github.com/mikeller/docker-betaflight-build ?

mikeller, 2016-11-13:
Fixed in https://github.com/mikeller/docker-betaflight-build/commit/e390d9c0e5269a18e41450265067e11fc641477c

closed 2016-11-13 by andymac4182