betaflight/betaflight #1541: Can't Flash Betaflight 3 to CC3D Atom

UltiFix, 2016-11-12:
I have tried it with the FTDI adapter and the main port. It just says failed to open serial port

Is it possible for you guys to make a .opbl file like with 2.9? I dont mind if you remove some features like LED, telemetry, battery voltage and stuff.

AndersHoglund, 2016-11-12:
Locate the Boot Loader pads on the FC, not sure where they are on an Atom. Short them by a solder blob or a wire..
Connect FTDI-USB to main port and PC, make sure the FC got power. Start BFC and select the "Firmware Flasher" tab, Select FC (1), select firmware (2), select "No reboot seq" (3), select "Manual Baudrate 115200" (4). Load firmware (5) and finally hit "Flash Firmware" button (6). Done. 
Remove the BL wire and connect USB. Make sure you have STM VCP drivers installed. Now BFC should connect just fine. Please make sure you use BFC 1.8.4 together with BF 3.0.1 firmware, latest release versions.

[image]

There are still OPBL variants available in all releases, but I would not use those.

UltiFix, 2016-11-13:
Thanks, it ended up working after some fiddling.

First off you have to have rx to tx and tx to rx. And you can't have the fc plugged into your pc. Even if you choose the port of the ftdi. It works if you only have the fc plugged into the usb port enough to get power but not get signal. Weird.

Whatever, I got it working and it flies. Thank you!

UltiFix, 2016-11-13:
Just curious where is the OPBL varient for cc3d?

AndersHoglund, 2016-11-13:
Of course you have to connect everything correctly. 
The OP Boot Loader takes up too much memory space. Therefore the betaflight OPBL variant has been stripped down to fit and is functionally crippled, no BF-PidC etc etc. On the github release page it is in the binaries.zip file.

UltiFix, 2016-11-13:
Oh OK, thanks!

closed 2016-11-13 by UltiFix