betaflight/betaflight #1542: QX90 Betaflight and Flysky FS-I6 Possible?

m88tow, 2016-11-12:
Well I have flashed my board with Betaflight instead of the cleanflight since so many said that it would fly much better. That was a process in and of itself to get it working and flashed properly. Now my next question is that I bought this thing to still fly even when its cold now that its getting close to winter so I can use it primarily indoors. I purchased it with the Flysky FS-I6 already bound to it. Now that I have updated this thing to betaflight I can't get the RX settings figured out in betaflight. I've searched high and low with no help yet. Is this RX locked out or something like that? Thanks in advance.

m88tow, 2016-11-12:
Never mind I got that part figured out.

I got everything working on the laptop but nothing works on the quad itself. The goofy thing is when I first connect into beta flight all I see on the simulated quad is 3 arms only 2 of which have props on them. Something is really goofy here

mikeller, 2016-11-13:
Looks like your Betaflight configurator installation is screwed up. Try uninstalling it and then reinstalling it.

m88tow, 2016-11-13:
Ok I will try that and report back, thanks for the tip.

m88tow, 2016-11-13:
Ok so I tried to uninstall it and reinstall it. I still have the same thing going on. I also downloaded cleanflight and got in there and it looks exactly the same. At this point I don't have any clue I have been messing with this stupid thing off and on for the better part of a day and a half now.

mikeller, 2016-11-13:
Can you please reopen this issue in the betaflight-configurator repo: https://github.com/betaflight/betaflight-configurator/issues

closed 2016-11-13 by mikeller