betaflight/betaflight #1548: RMDO dshot not working on build 706

brycedjohnson, 2016-11-13:
Supposed to be pin compable with SPRACING but doesn't appear to work.  SPRACING hex on dodo is working fine.

May need this in the target.h on RMDO that is in the spracing target.h?

#define USE_DSHOT
#define REMAP_TIM17_DMA

// UART1 TX uses DMA1_Channel4, which is also used by dshot on motor 4
#if defined(USE_UART1_TX_DMA) && defined(USE_DSHOT)
#undef USE_UART1_TX_DMA
#endif

borisbstyle, 2016-11-13:
Yeah it needs remapping indeed.

blckmn, 2016-11-15:
https://github.com/betaflight/betaflight/pull/1562 will sort this one out.

blckmn, 2016-11-15:
Please test and report back so this may be closed if resolved.

blckmn, 2016-11-16:
@brycedjohnson any luck on this one?

brycedjohnson, 2016-11-16:
Haven't had a chance to test yet... Just got home.  I'll do that right now.

brycedjohnson, 2016-11-16:
Yup worked for me.  Thanks for getting that in there.

Tested with build 714 which had up to commit 819925b4caadafe68049077d90bf235fd1fca15d.

closed 2016-11-16 by brycedjohnson