betaflight/betaflight #1552: naze32 只有在baseflight  刷2015-5的固件正常

jiekchenid, 2016-11-14:
。naze32 只有在baseflight  刷2015-5的固件正常，GPS，电调都正常。。。cleanflight  betaflight 刷完固件都不正常，都不能校准电调，也检测不到GPS,甚至刷他俩最新的固件，都无法校盘罗盘和水平，按钮都是灰色的。是什么原因，校准时电调时，通电，电机总是进入编程模式，是不是因为我的板子太老了。。刷betaflight 刷一个老的固件能校准水平和罗盘，但是校准不了电调，，油门最大，一通电，电机的响声不是校准模式，而是某个编程模式！！！！！GPS也检测不到。。。。。。但刷回baseflight 就一切正常。

blckmn, 2016-11-14:
Google translate (if anyone can translate better please do): 
naze32 only baseflight brush 2015-5 firmware normal, GPS, ESC are normal. . . cleanflight betaflight brushing firmware is not normal, you can not calibrate power transfer, can not be detected GPS, they even brush the latest firmware, it can not compass disc and school level, the button is gray. What is the reason, when the calibration ESC, electricity, motor always enter the programming mode, it is not because I am too old board. . Brush brush betaflight an old firmware can calibrate the level and compass, but can not calibrate the ESC ,, maximum throttle, a powered motor is not sound calibration mode, but a programming mode! ! ! ! ! GPS can not be detected. . . . . . But brushed back baseflight all normal.

metalnow, 2016-11-14:
briefly speaking, he said his naze32 only can run baseflight v. 2015-5. cleanflight and betaflight won't work on his machine.

jiekchenid, 2016-11-15:
现在的问题时，我可以刷betaflight固件了，但是刷完betaflight_3.0.1_NAZE这个版本后，，固件不支持mag baro 在配置页面这两项都选了，，，我的硬件是naze32 rev5 10dof

jiekchenid, 2016-11-15:
刷了2.9.1固件后，，一切都正常了

metalnow, 2016-11-15:
does anyone have naze32 rev5 10dof board to test mag and baro in 3.0.1 ? @jiekchenid  said those are only working in 2.9.1. please help.. thanks.

blckmn, 2016-11-15:
In 3.0.1 they were disabled due to space constraints. 3.1 is introducing the baro back, but still not enough for the mag yet.

metalnow, 2016-11-15:
@blckmn got it. thank you.

@jiekchenid 3.0.1 有空間不足的情況，所以不支持 mag, baro. 3.1 的 baro 會有，但mag可能還沒有辦法。

jiekchenid, 2016-11-15:
thanks everyone!

closed 2016-11-16 by jiekchenid