betaflight/betaflight #1553: Serial port for SerialRX will not get activated anymor in actual master

MJ666, 2016-11-14:
The serial port will not set to SerialRX anymore (ports page) but the serialRX feature is still active (configuration page). Here are the assignments for the AlienFligthF4 wich are affected.:
https://github.com/betaflight/betaflight/blob/master/src/main/target/ALIENFLIGHTF4/target.h#L165-L166

Here is the part when it should be activated:

https://github.com/betaflight/betaflight/blob/master/src/main/fc/config.c#L769-L774

I may need to debug since at the first view it looks good to me. May be someone has already a clue?

mikeller, 2016-11-14:
Potentially caused by #1527. Any ideas, @blckmn?

blckmn, 2016-11-14:
Ah. I think we just need to add that the port is enabled. I'll take a look. 

Note this tested ok on a bluejay- which I've now moved to using serial (uart6 / sbus) as default not PPM

MJ666, 2016-11-15:
I found the source of the problem. Likely some more targets are affected. The useage of the SERIALRX_UART has changed by the commit below and corrected some odd behaviour from the past. 

https://github.com/betaflight/betaflight/commit/0ebaef175ecc88c6008d917b31e675c58ef530f5#diff-d7ecdbee607f4619a513374bcc4fd179

The real serial port used for the AlienFligth F4 target for the RX is on UART2. In the past you needed to specify one port higher to get it work properly. Please see the two related lines. 

https://github.com/betaflight/betaflight/blob/master/src/main/target/ALIENFLIGHTF4/target.h#L154
https://github.com/betaflight/betaflight/blob/master/src/main/target/ALIENFLIGHTF4/target.h#L166

If the SERIAL_PORT_USART3 will be changed to SERIAL_PORT_USART2 everything is working correctly for the AlienFlight F4 now. My guess is this is a problem for all VCP targets where VCP is shifting all serial ports up by one in the list. How do we find all affected targets? Likely no one have all of them for checking.

mikeller, 2016-11-15:
Good find, @mj666. If it affects all VCP targets then finding them is easy, just grep for `USE_VCP` in `target.h`.(it's almost all targets anyway). Fixing all of them will be a bit of work, but in my opinion it's worth it, because it reduces the mix of 0 based and 1 based counting of the serial ports.

blckmn, 2016-11-16:
I think we definitely want to clean up the targets. I'll have a look tonight, and raise a PR

MJ666, 2016-11-16:
Have started working on this.

MJ666, 2016-11-16:
PR #1577. hope I catched everything.

MJ666, 2016-11-16:
Fix is merged.

closed 2016-11-16 by MJ666