betaflight/betaflight #1554: Baro betaflight_3.1.0-709  8 8 32000 Dshot150 -two motors on the left do not work

Cthutquz, 2016-11-14:
KKNGF4:Flash:6000:No Baro betaflight_3.1.0-709-32c87d2
Racerstar RS30A V2 30A Blheli_S ESC OPTO 2-4S Support Oneshot42 Multishot ( G_H_30_REV16_43.HEX )
8 8 32000 Dshot300 -work
8 8 32000 Dshot150 -**two motors on the left do not work**, two motors are working right
8 4 32000 Dshot150 all are working properly

lunohod, 2016-11-14:
I've got the same issue with DTFc (@ build 702) + Aikon 30a V1.
Dshot600, 8k/4k: right motors work fine, left motors do not spin up. 
Dshot300, 8k/4k: All motors spin up, left motors occasionally desync (the copter jumps in the air).

Cheredanine, 2016-11-15:
@lunohod,
You issue is typical of cap smoothing on the signal input of the escs, find the right cap and remove it, it will be fine for dshot, although no guarantees your PWM will be smooth

OPs issue may be different as faster protocol works

@Cthutquz 
Have you tried reflashing with a different build? Been running since first Dev releases, had the issue you were describing once, reflashed with a later build and all good. Not sure if it was the build or the reflash that fixed

Jahnkeanater, 2016-11-16:
I am having the same issue as OP. Motors 3 & 4 will not arm at 8k with Dshot150. This could be as simple as the capacitor but I wonder if maybe it is a timing issue with 8khz and Dshot150?
I tested Dshot300 and had no issues after 1 flight.
Revo betaflight_3.1.0-714-819925b
BusyBee2 ESCs

borisbstyle, 2016-12-19:
Closing this one as it looks like ESC issues

closed 2016-12-19 by borisbstyle