betaflight/betaflight #1563: Full Throttle Backflip at ARM with HGLRC FC

AJSNetworks, 2016-11-15:
Hello,
I recently rebuilt a quad from a RTFQ F3 board to HGLRC F3 AIO V3.1.

I copied the working config over from the old fc to the new fc and during arming test without rotors I noticed very aggressive Yaw oscillation and dropped yaw from 35 to 10 which eliminated the problem.

I put the rotors on, powered the quad up and upon arm with zero throttle it rapidly went full throttle on M1 and M3 causing the copter back flip uncontrollably.

[lastknowgoodconfig.bf3.0.1.txt](https://github.com/betaflight/betaflight/files/590962/lastknowgoodconfig.bf3.0.1.txt)
[blackbox_log_2016-11-14_192041.TXT](https://github.com/betaflight/betaflight/files/590963/blackbox_log_2016-11-14_192041.TXT)

Ive attached the blackbox and the config.

andre92, 2016-11-15:
could you verify correct board alignment and motor wiring order? Blackbox indicates the copter backflipped, yet it gave full throttle to back motors, which should stop it. Since that only accelerated the backflip, that indicates either the motors aren't wired correctly (front and back swapped), or the board alignment needs to be set properly (enable ACC and verify correct position of 3d model)
Make sure you remove props when checking.
I probably won't need to mention correct motor direction and propellers since you only changed the FC.

AJSNetworks, 2016-11-16:
End user issue. I configured motors to APMs motor positions rather than the CF/BF positions, hence the eradicate behavior.

Thank you reviewing my issue.

closed 2016-11-16 by AJSNetworks