betaflight/betaflight #1566: Resource Remappin dont work

ps915, 2016-11-15:
I tryied to change the Beeper Out Pin, but with no success.

I tryed these commands with no luck:

resource BEEPER B5
resource BEEPER B05
resource BEEPER OUT B5
resource BEEPER OUT B05

I also did not forget to hit save

(Screenshot will show B04)
[image]

AndersHoglund, 2016-11-15:
Resource remapping will be available in 3.1 You seem to be on 3.0.0

ps915, 2016-11-15:
Where can i get V3.1
tested it on the latest 3.0.1 Also dont work
[image]

blckmn, 2016-11-15:
Latest builds are available here:
http://andwho.sytes.net:8080/job/BorisB_BetaFlight/lastSuccessfulBuild/artifact/obj/

closed 2016-11-15 by blckmn