betaflight/betaflight #1568: Disappearing OSD on Omnibus F3

sgcdialler, 2016-11-15:
The OSD disappears on the Omnibus F3, corresponding to large power spikes (associated with hard/fast maneuvers with active-breaking enabled on ESCs). This appears to be the same issue that occurred with Matek HUBOSD PDB boards. Both the HUBOSD and the Omnibus use the MAX7456 chip to provide the OSD function. HUBOSD runs the STOSD firmware, which fixed the issue in v3.2. Now, instead of disappearing, the OSD on a HUBOSD board will flicker under heavy spikes but not disappear permanently.

nathantsoi, 2016-11-15:
What kind of escs are you using?

sgcdialler, 2016-11-15:
Littlebee 30A-S BLHeli_S

jflyper, 2016-11-18:
@sgcdialler How are you powering your cam?

sgcdialler, 2016-11-19:
Cam and VTX are both powered by 12V reg on the PDB. Just the overlay disappears, the video stream is fine.

jflyper, 2016-11-19:
I believe it is not a software issue.

(3) If you don't have ground connection between CAM-OSD and OSD-VTX, please add them.
(4) CAM going undervolt produces flickering overlay. Review your power system.

If these don't fix your problem, please come to the
[RCG OMNIBUS F3 thread](https://www.rcgroups.com/forums/showthread.php?t=2711617&page=50#post36233251).

borisbstyle, 2016-12-19:
Closing due to non software issue. Discussion can always continue

closed 2016-12-19 by borisbstyle

bob9, 2017-04-02:
I have this same issue did you get a result?

nathantsoi, 2017-04-02:
If it's power related, you could try an LC filter on your camera power
supply. If your ESCs are noisy (like littlebees), add a capacitor where
your power lead connects to the ESC.

bob9, 2017-04-02:
I think your right.  I've got an LC filter already. After some more reading
I think a lack of common ground between FC vtx and camera may be the issue.

I added the common ground and I'll see how it goes next flight.

If that fails me I'll but the cap on.

Thanks for you help..

On my phone...

benchamin23, 2017-07-23:
I have the same issue on my wing. I use the Wraith32, although I have powered my VTX directly by battery with an LC filter and my cam is powered by my VTX.