betaflight/betaflight #1572: EMAX Skyline32 + OSD - BFOSD Compatibility

Vadhvis, 2016-11-16:
Any details about BFOSD compatibility with this board?

mikeller, 2016-11-16:
According to this the on board OSD runs minimOSD, i.e. it comes with its own MPU:

https://oscarliang.com/skyline32-osd-f1-f3/

This means it does not run BFOSD. But there are several projects under way to better integrate minimOSD  with Betaflight.

Vadhvis, 2016-11-16:
According to the [official site](http://www.emaxmodel.com/fpv-controllers/emax-skyline32-osd-flight-controller-acro.html) it runs MW_OSD

Does this change anything?

mikeller, 2016-11-16:
You're right. The OSD firmware is actually `MW_OSD`, the hardware is called minimOSD. So this is the same really, it's not the main MCU that is running Betaflight that is driving the OSD hardware.

Vadhvis, 2016-11-16:
I see, thank you very much

closed 2016-11-16 by Vadhvis