betaflight/betaflight #1578: SPRacing F3 EVO BF3.1 #715 DShot problem

jorgemigalves76, 2016-11-16:
After select DShot 150, 300 or 600 and saving config, the FC dont boot anymore. Need to put in dfu and reflash. Anyone with same issue? Thanks.

Jahnkeanater, 2016-11-18:
SPRACINGF3EVO - Locked out Comm port fixed (build 708?). Must move MOTOR 4 to new pin assignment (CLI = resource MOTOR 4 A06). Then solde ESC for motor #4 to motor output #5, fixes DMA conflict with motor outputs 2 and 4. See: https://github.com/betaflight/betaflight/wiki/Betaflight-specific-CLI-commands#resource-remapping-command-v31

blckmn, 2016-11-18:
Yes - this issue has been fixed in later builds.

closed 2016-11-18 by blckmn

Tdogb, 2017-12-21:
This issue has appeared in 3.2.3

jflyper, 2017-12-21:
@Tdogb Your issue is most likely a new one. Please open a new issue ticket following the instruction in the pre-formatted text.