betaflight/betaflight #1581: LED lights not working on REVO

trojanGoat, 2016-11-16:
I'm following a thread on RCGROUPS.  People have suggested that the led signal has been moved to motor 5.  I've gone ahead and connected the led to motor5 and still no lights are working.  Not sure if this is a betaflight or hardware issue.

https://m.rcgroups.com/forums/showthread.php?t=2731917&page=39#post36213798

AndersHoglund, 2016-11-17:
LED-strip on my REVO works fine using motor 5 output. Just select LED-strip feature and then some LED pattern. 

    # version
    # Betaflight/REVO 3.1.0 Nov 17 2016 / 17:33:20 (135231f)
    
    # feature
    Enabled: FAILSAFE LED_STRIP BLACKBOX 
    
    # led
    led 0 4,15::CO:2
    led 1 5,15::CO:2
    led 2 6,15::CO:2
    led 3 7,15::CO:2
    led 4 8,15::CO:2
    led 5 9,15::CO:2
    led 6 10,15::CO:2
    led 7 11,15::CO:2

Have you done the required "diode in series"-mod, lowering VCC to LED-strip?

mikeller, 2016-11-18:
@AndersHoglund Is there documentation on this "diode in series"-mod? I wasn't aware that such a thing is needed, but I've had irregular behaviour in the past that I put down to voltage problems.

trojanGoat, 2016-11-18:
No I haven't done the vcc to led strip.  I didn't realize that was required.  Can you send a link?

AndersHoglund, 2016-11-18:
Since some/most of our FCs are only driving high output signals to 3.3V, there is a problem with LED-strips requiring a high level of at least 0.7*VCC (0.7*5=3.5V). Lowering VCC with a diode in series is the easiest fix. Might also work with a pullup or a transistor buffer,  have not tried that. My Adafruit Neopixel is connected with a diode in series. Built into the cable. 
This a well known problem and fix, even covered in da dox: https://github.com/cleanflight/cleanflight/blob/master/docs/LedStrip.md RTFM ;-)

closed 2016-11-19 by blckmn