betaflight/betaflight #1582: BMP180 fails to connect to Flip32 Omnibus F4

Loswave, 2016-11-17:
With the sensor connected to the i2c bus on the device it will register 2 i2c errors on boot, but will not see the barometer. I have verified the barometer works fine with an arduino.

I'm assuming this is an i2c issue with the flip32 omnibus f4.

I am running betaflight 3.0.1.

I have verified proper wiring.

I have an oscilloscope with bus decoding built in and would be happy to do any testing that might help.

nathantsoi, 2016-11-17:
could you post a picture? i'm pretty sure the omnibusf4 doesn't have a
baro, the omnibusf4v1 has a bmp280 on spi. the binary for the v1 isn't in
the main repo yet, i'm working on hardware detection if you want to help
out /w testing?
https://github.com/nathantsoi/betaflight/tree/target/obf4-hw-rev

Loswave, 2016-11-17:
Shoot, I'm sorry, I was trying to make sure I included all details then missed a big one. This is an external sensor I was attempting to connect. I know it was supposed to work on other boards and I just assumed it would on this as well, which may well be my mistake.

I know I read it was doable on old naze acro boards so for some reason I just assumed it was universal without thinking it through properly. So if this is not a done thing I apologize and you can close this issue up.

On the other hand if it's doable that would be fantastic, mainly just cause I like altitude on my OSD.

I wasn't aware there was another revision of the omnibusf4, so I'm reasonably sure this is not that. I also did not see a barometer onboard. Looks like I should get a v1  when I can find one!

In any event I'd be happy to test/help if needed.

[image]

jflyper, 2016-11-17:
@Loswave I2C support on OMNIBUS F4 is planned in the 3.1.

@nathantsoi 
Two things regarding the I2C support on OMNIBUS F4.

1. Pin assignment
Current target.h specifies I2CDEV_1 which I believe goes to RC_CH1 and RC_CH2, but pin map suggest that I2C is shared with UART3 which requires I2CDEV_2. Am I confused in some way? Ofcourse, if i/o resource allocation will include this kind of remapping, then we don't have to worry about it.

2. Baros and Mags on I2C
I think we should add some more baro and mag support; we only have MS5611 atm.

```
#define BARO
#define USE_BARO_MS5611
#define USE_BARO_BMP085
#define USE_BARO_BMP280

#define MAG
#define USE_MAG_AK8975
#define USE_MAG_HMC5883
```

borisbstyle, 2016-12-20:
@jflyper Is this implemented in the current master?

jflyper, 2016-12-20:
@borisbstyle @nathantsoi 

It's not in yet.

I'm thinking of two configuration options, but they both have a trade off.
1. Use I2CDEV_2 sacrificing UART3.
With current master, i2c seem to have priority over uart, so once the I2CDEV_2 is configured, UART3 is effectively suppressed its usage.

2. Use PWM6 (Motor 6) and S6_IN (small pad) for I2CDEV_3.
This requires breakout wire for S6_IN.

I'm modding i2c codes so that i2c can also be configurable with resource command, but I'm not sure if it will be ready for 3.1. A pragmatic solution, if I2C support was to be included in 3.1, would be to create a config variable to specify active bus(es).

Any thoughts?

jflyper, 2017-07-26:
@Loswave 
- Configurable I2C has been implemented and merged.
- Configurable barometer device has been implemented and merged.

As BMP180 is a drop-in replacement for BMP085, you can now add `USE_BARO_BMP085` to see if your BMP180 will work or not.

avidtech, 2017-08-13:
@jflyper 

I attempted to get my BMP180 working, I did get my i2c bus functioning with my HMC5883l (reference #3605 for everyone else following the issues). I had no such luck using any of the data here to get the baro working at all.  After trying the default config, I have set the following in the CLI, `set baro_i2c_address = 119` following the product datasheet's information for  addressing to no avail. All of this is on the OBF4, same version as @Loswave has posted.

Apologies if I left anything out, haven't slept in two days.

Thank you!

jflyper, 2017-08-13:
@avidtech My apologies. BMP085/180 was not configured in the OMNIBUSF4 target. I added the support and confirmed it working. Please wait for RC4 or wait few hours and get the latest dev build to try.

avidtech, 2017-08-13:
Thanks again @jflyper for your efforts!

avidtech, 2017-08-27:
I'm back again, 

After a good amount of work and testing, I still can't get my BMP180, OR my BMP280 to be recognized by the FC. While the external HMC5883 mag is working great on the FC, and all(HMC5883, BMP180, BMP280) peripherals were just re-tested on my RPi I2C bus (all passed), I double checked all of the contacts and harness, everything is spec+. While using both my single modulae harness for one of the devices, and my two module harness, the mag works every time, but the baro does not, but ONLY on the FC with no I2C errors. Everything tests fine on my RPi. As can be seen in the attached images, the barometer reports an address of 118/0x76.

I couldn't find any known issues with using custom I2C addresses. Is that potentially an issue?

I'm at a loss here, guys.

Target: OmnibusF4 (v3)
Build: Latest

[Current config](https://drive.google.com/open?id=0Bx9_NkoiR0PIdWRkb1ZoTmtsOUU)

[I2C addressing report](https://drive.google.com/open?id=0Bx9_NkoiR0PIc3g5SmxmLXJvSk0)

Thanks again, let me know if I need to get any more info.

avidtech, 2017-09-03:
@jflyper ,
After numerous days tinkering, I've got nothing. To restate, the TL;DR is I2C Mag works, baro does not (BMP280 or BMP180). Thoughts? Ideas? I'd really like to have this sensor operational on my craft.

Thanks again friends

jflyper, 2017-09-03:
@avidtech Let me see your cli diff at this time.

avidtech, 2017-09-03:
Yep, already have the pin resources set. Honestly, I thought it was my fault at first, that I missed something or fat-fingered something. I even reduced my bus cable length to less than 1" to test, but both sensors work great with the Pi. 
For good measure:
`# Betaflight / OMNIBUSF4 (OBF4) 3.2.0 Aug 28 2017 / 01:32:16 (ee4827246)

# name

# resources
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11

# mixer
mixer HEX6X

# servo

# servo mix

# feature
feature MOTOR_STOP
feature GPS

# beeper

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set mag_hardware = HMC5883
set mag_declination = -3400
set baro_i2c_address = 119
set baro_hardware = BMP280
set rssi_channel = 9
set serialrx_provider = SPEK2048
set motor_pwm_protocol = DSHOT600
set gps_auto_baud = ON
set deadband = 2
set yaw_deadband = 2

# profile
profile 0

# rateprofile
rateprofile 0`

jflyper, 2017-09-03:
Your `baro_i2c_address` is 119. Is this intensional? Your device probe shows it's 118 (0x76).

avidtech, 2017-09-03:
Yes. In my longer post without the diff out, the original address was 0x76. I changed the solder pads to change the address to 0x77 and verified it with the Pi.

avidtech, 2017-09-03:
I'll be back home in a few hours. If there's any debugging proceedure or testing you'd like me to help with, I'm all for it. If there is a change made in the current master I'll recompile and test right away.

Thanks for the help

jflyper, 2017-09-03:
How many I2C errors do you get? Can you run a test with

1. Mag and Baro connected
2. Mag and Baro both disconnected

avidtech, 2017-09-04:
@jflyper ,
Sorry for the delay. 
Both plugged in, no I2C errors. Unplugged on a fresh boot, always 4 I2C errors. Unplugging while running, error count ticks up forever.

avidtech, 2017-09-26:
So guys,

Any thoughts on this or am I stuck finding a new FC? The baro is mission critical for my application. It's been a while, so I'll recompile and try again. It's very frustrating since the mag works but the baro doesn't, on the same physical bus line.

Thanks again, I'll help however I can, just ask.

jflyper, 2017-09-26:
@avidtech I'm looking forward to hearing the new results.

avidtech, 2017-09-26:
I'll check it out in the afternoon. I've been busy with building my supercapacitor pulse welder B)

Will advise,

Thanks again.

wx4cb, 2017-12-02:
same issue here, brand new omnibus F4 nano (bmp280 - flip32 version i guess), it's using the lux4osd target and not able to get any form of baro detected

I've tried the above cli comments as an experiment. is there another target i can try with this board? this is the only 20x20 board I can find that has a baro.

mikeller, 2017-12-02:
@wx4cb: I guess this is a case for opening a support request with your supplier or the manufacturer. Their instructions state that you have to install the `OMNIBUS` target - which is clearly wrong, since the OMNIBUS is an F3 based board. Ask them to fix their documentation (http://www.readytoflyquads.com/flip-32-f4-omnibus-nano) to properly state which target to use, and if it needs a custom target, provide a target definition to the Betaflight project so that it can be added.

wx4cb, 2017-12-02:
yea i just emailed them to see what they can do. I got this board because it's the only one i could find thats 20x20 with a baro - going into a small plane/wing

avidtech, 2017-12-02:
Curious, I haven't gotten my old F4 Omnibus baro working yet, but that's assuming the manu is correct in specifying it's an F4 target board. 

@mikeller was "F3" a typo or could this be related to all of the units with the F4 specification? 
@wx4cb I am curious to your results, even though our boards are slightly different, I STILL to this date have had zero luck getting a bussed barometer to work at all.

mikeller, 2017-12-02:
@avidtech: I don't know. And since there are different variants of the OMNIBUSF4, unless they fix their documentation it's just guesswork on what target will work...

avidtech, 2017-12-02:
@mikeller fair enough, I thought you may have had some secret spy information B)

I have shelved my F4 powered hex for this issue. Maybe I will come back to it and start poking around again. I have no problem pointing and shooting with different targets for testing, so to speak.

wx4cb, 2017-12-02:
@avidtech you and me both alex, I sent the following email off to them to see if they have any thoughts:

@mikeller Thanks for giving me that info, hopefully they understand it too :D

Hi

I ordered an omnibus F4 nano from you guys this past week and i
received it today.

Product link: http://www.readytoflyquads.com/flip-32-f4-omnibus-nano

however, upon plugging it into betaflight, the barometer isn't being recognised.

According to your documentation on the site, (which is allt he
documentation there is other than a pinout diagram) you state the
omnibus F4 target, however, this board came flashed with the LuxF4OSD
target (CLI and configurator issue below).

So my question is, which target is correct to get the barometer
working. That was the reason why i got this board as I needed a small
20x20 board with a baro for a small plane im designing.

I posted this question on the betaflight github ( https://github.com/betaflight/betaflight/issues/1582#issuecomment-348726757 )and they responded:

mikeller commented 26 minutes ago

@wx4cb: I guess this is a case for opening a support request with your
supplier or the manufacturer. Their instructions state that you have
to install the OMNIBUS target - which is clearly wrong, since the
OMNIBUS is an F3 based board. Ask them to fix their documentation
(http://www.readytoflyquads.com/flip-32-f4-omnibus-nano) to properly
state which target to use, and if it needs a custom target, provide a
target definition to the Betaflight project so that it can be added.

*** On the site: ***

Target "OMNIBUS F4" to update the firmware.

*** Ass seen by the betaflight configurator

2017-12-02 @ 18:28:18 -- Serial port successfully opened with ID: 47
2017-12-02 @ 18:28:18 -- MultiWii API version: 1.36.0
2017-12-02 @ 18:28:18 -- Flight controller info, identifier: BTFL,
version: 3.2.2
2017-12-02 @ 18:28:18 -- Running firmware released on: Nov 11 2017 00:34:30
2017-12-02 @ 18:28:18 -- Board: LUX4, version: 0
2017-12-02 @ 18:28:18 -- Unique device ID: 0x2e004e3036510832333439
2017-12-02 @ 18:28:18 -- Craft name:

And CLI Dump

# Status
System Uptime: 34 seconds
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000
Stack size: 2048, Stack address: 0x10010000
I2C Errors: 0, config size: 2085, max available config: 16384
CPU:4%, cycle time: 126, GYRO rate: 7936, RX rate: 49, System rate: 9
Arming disable flags: ANGLE CLI

# version
# Betaflight / LUXF4OSD (LUX4) 3.2.2 Nov 11 2017 / 00:34:30
(77782e602) MSP API: 1.36

jflyper, 2017-12-03:
@wx4cb

**LUXF4OSD target does not support onboard SPI connected BMP280.**

The Omnibus F4 Nano should be flashed with *OMNIBUSF4SD* target. I suspect that RTFQ has flashed it with LUXF4OSD because someone told them that LUXF4OSD will enable onboard flash device _if users solder a part onto the board_.

wx4cb, 2017-12-03:
@jflyper yes this has adataflash chip on it. does the omnibusf4sd target support the dataflash chip ?

jflyper, 2017-12-03:
Did it come with one soldered?

wx4cb, 2017-12-03:
i ordered it with one yes and betaflight sees the dataflash chip

jflyper, 2017-12-03:
Oh, I see, they have an option for the flash.

Unfortunately, the flash and the baro do not co-exist in any of the existing Omnibus F4-like target.

Let me give it a quick thoughts. Gimme 15min. Hang on.

digitalentity, 2017-12-03:
I have a feeling that they might. We don't use DMA for accessing dataflash, writes are 100% synchronous, therefore we can probably use baro+ms26p16 on the same bus.

wx4cb, 2017-12-03:
ok i tried the omnibusf4SD image and the baro works but obviously no dataflash chip.

What can i do to help get either/or working?

would a cli dump of each target be of use ?

jflyper, 2017-12-03:
@wx4cb Hang on. I'm working on it.

jflyper, 2017-12-03:
@wx4cb 
Try with this firmware, full-chip erase flashing.

[betaflight_3.3.0_LUXF4OSD-with_SPI_BMP280-00.hex.zip](https://github.com/betaflight/betaflight/files/1524398/betaflight_3.3.0_LUXF4OSD-with_SPI_BMP280-00.hex.zip)

If it doesn't work, let me see the result of CLI commands `get baro` and `resource list`.

wx4cb, 2017-12-03:
Dataflash is there, but no baro yet. here's the results of the 2 commands you needed. 

Just as an FYI, I tried setting the baro_bustype to spi and that didnt work. but just out of curiosity, i used the omnibusf4sd target and the barometer worked, but obviously no DF chip. If any of that helps

# get baro
baro_bustype = I2C
Allowed values: NONE, I2C, SPI, SLAVE

baro_spi_device = 0
Allowed range: 0 - 5

baro_i2c_device = 2
Allowed range: 0 - 5

baro_i2c_address = 0
Allowed range: 0 - 119

baro_hardware = AUTO
Allowed values: AUTO, NONE, BMP085, MS5611, BMP280

baro_tab_size = 21
Allowed range: 0 - 48

baro_noise_lpf = 600
Allowed range: 0 - 1000

baro_cf_vel = 985
Allowed range: 0 - 1000

baro_cf_alt = 965
Allowed range: 0 - 1000

# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: MPU_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B03: SPI_PREINIT
B04: BEEPER
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: MPU_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
D02: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

Use: 'resource' to see how to change resources.

jflyper, 2017-12-03:
Okay, do these CLI
```
set baro_bustype = SPI
set baro_spi_device = 3
resource baro_cs b3
```

wx4cb, 2017-12-03:
hehehe. i was way ahead of you - lol.... I flashed the f4sd target on and did the same commands and was just typing in those above commands when you posted it.. VOILA I see barometer in the sensors thab.... 

appreciate it. i'll now have to sit down and figure out setting it up in a plane (i only really need it for variometer and the osd - hence the cheap 20x20 board :D

wx4cb, 2017-12-03:
here's the pastebin of those commands from before: https://pastebin.com/X6WNDPp1

jflyper, 2017-12-03:
@wx4cb I will probably do some more investigation and make `OMNIBUSF4` target compatible with Nano V2. So, please pay attention to board page on the BF wiki for future releases.

wx4cb, 2017-12-03:
also while you're at it, and im digressing from this topic, but this board is supposed to have an osd, i'll have to investigate more, but i may take a look at the other (F4SD) target to see if the osd works on that target.

I really appreciate the help you gave tonight :D one step closer to my project goal heh

jflyper, 2017-12-03:
It's 10 in the Sunday morning here :)

wx4cb, 2017-12-03:
hehehe 8pm sat night here and my son is bitching he wants some food LOL

avidtech, 2017-12-27:
@jflyper ,

Wow, sorry for taking forever to get back, I'm currently using the newest stable build offered as of this post's date. I am still getting the same behavior as mentioned in the past, mag works flawlessly, zero recognition of the BMP280 on the i2c bus. The two sensors are basically attached at the same point in the bus wiring, and this mini sensor cluster is tested working with a raspberry pi as it is physically configured without any issues. I am still at a loss here, I'm going to play around with some of the CLI hardware configurations and see if any changes are noted. I am unsure of the behavior of a few of these variables with BF, notably `baro_i2c_device` and `baro_tab_size` (frame size or other data-link layer parameter?). I'm dead in the water with ideas here, especially since the baro/mag combo works great on other decoding devices just fine as it is physically arranged, and the mag is showing great data in the configurator/OSD. Again, I have no i2c errors, too.

If you'd like me to dump anything for this purpose, please let me know. I'll be at my machine trading crypto and generally just fiddling with this thing for a while today.

Thanks everyone, for all of your hard work and dedication!

-A

jflyper, 2017-12-27:
@avidtech So, this is a BMP280 connected over I2C, in parallel to a mag?

Can I see CLI "`diff`", "`resource list`" and "`set baro`" results?

avidtech, 2017-12-27:
Yes, sir. I'll collect that information right now.
`# set baro
baro_bustype = I2C
Allowed values: NONE, I2C, SPI

baro_spi_device = 0
Allowed range: 0 - 5

baro_i2c_device = 2
Allowed range: 0 - 5

baro_i2c_address = 0
Allowed range: 0 - 119

baro_hardware = BMP280
Allowed values: AUTO, NONE, BMP085, MS5611, BMP280

baro_tab_size = 21
Allowed range: 0 - 48

baro_noise_lpf = 600
Allowed range: 0 - 1000

baro_cf_vel = 985
Allowed range: 0 - 1000

baro_cf_alt = 965
Allowed range: 0 - 1000`

`# cli diff
Unknown command, try 'help'
# diff

# version
# Betaflight / OMNIBUSF4 (OBF4) 3.2.3 Dec 11 2017 / 07:56:31 (cb962eda1) MSP API: 1.36

# name

# resources
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11

# mixer
mixer HEX6X

# servo

# servo mix

# feature
feature MOTOR_STOP
feature GPS

# beeper

# map

# serial
serial 0 64 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set mag_hardware = AUTO
set mag_declination = -3400
set baro_hardware = BMP280
set serialrx_provider = SPEK2048
set blackbox_device = NONE
set motor_pwm_protocol = DSHOT600
set gps_provider = UBLOX

# profile
profile 0

# rateprofile
rateprofile 0
`

Nothing terribly exciting to see. This is pretty much a fresh flash without much other configuration, just a focus on data acquisition configuration.

avidtech, 2017-12-27:
Settting baro address to 118 or 119 yeilds no information. I don't remember offhand, I believe I have the sensor SMD solder bridge set to 118.

jflyper, 2017-12-27:
@avidtech This will take a while ... I have to setup a test env, but my bench is under re-structuring...

avidtech, 2017-12-27:
I understand, if there is anything you'd like me to do for you to help, please feel free to ask. I wish I could have been involved with BF at an earlier point since there is so much to catch up on. That being said, I am a "hardware guy", so take that with a grain of sand xD

Thank you for everything.

avidtech, 2018-01-23:
Update, @jflyper ,
I still haven't gotten anywhere. I am going to flash the *wrong* target as mentioned, the SD version, and see if I can manage without the dataflash. I don't *really* need the dataflash, but I'd prefer to have both.

Thanks again.

avidtech, 2018-03-22:
Ok, @jflyper , my new MS5611 baro DOES work over the I2C bus on the omnibus V3 board (and the pro, but this is not of much use seeing as that has the SPI BMP280) with the same settings (except for the manual hardware selection) as I ran with the BMP280. I re-tested the two BMP280/GYBMEP boards on my raspi, the addresses are correct and standard (on one, there is a solder bridge for selecting the two primary output addresses, tried both), so I really have no idea what's going on there, seeing as I am not a software dev. :) Maybe it's something I overlooked in the source, when I was diagnosing? Maybe it's the refenced addresses in the source that don't jive, I did not dig that deep, and you fellas know it much better than I do for sure.

jflyper, 2018-03-22:
@avidtech I suspect level-shifters on your bob is what is preventing the FC to talk to the BMP280 chip. If you are still interested, you can try the naked (non-shifted) bob such as the one available under the name GY-BMP280-3.3 or equivalent, available at $1 a piece.

avidtech, 2018-03-22:
@jflyper , you're talking about the resistors? I believe those should reduce the dynamic range of the I/O, but I haven't looked at it in great detail lately. I thought my board had a 5v pull-up for the output. This could be valuable for others to know for sure, at this point though, I finally have two Omnibus FC's with baros and mags (So happy! lol), the V3 is supporting an original HMC5883L and MS5611 on I2C, and the PRO V3 is supporting the SPI BMP280 with original HMC5883L on I2C with no issues as stated. 

Let me check, I swear this BMP280 board was set up for 5v logic, and the GPIO on the raspi was happy with the signal too, but that's probably fine with 3.3v logic. I have to write some certificates up for some clients then I will check and report tonight.

jflyper, 2018-03-22:
The level shifter in question is composed of one n-channel MOS FET for each channel (SDA and SCL), and designed to interface host side (Vcc) and BMP side (3.3V). I have to scope it to be definite.

avidtech, 2018-03-22:
Ah ok, so the logic from the BMP will feed into the gate then, correct? I'll look up the IC and get the pinout.

avidtech, 2018-03-22:
Well, either the last one I have left is too damaged, or it just won't talk to the board. HMy only working scope is a USB piece of junk, but it does work. Clock is being passed in ok, but the SDA line is just showing a steady 3.3v, so I guess that's a good clue here. I think you're right, but I am not sure why that baro was advertised as a full range 5v logic board, and there were other 3.3v spec specifically labeled that way. Oh well, the new MS5611 signal looks beautiful, really low electrical noise, and the sensor itself is really low noise too, so I cannot complain with what I have.

Related, I have been running into what seems to be an increase in improperly marketed parts, especially in logic I/O (TTL/CAN) so I am not shocked about this.

jflyper, 2018-03-22:
Ah, I'm sorry, I should have been more specific. I know exactly how the circuit is made. What I'm wondering is if the level shifter designed and tuned to shift between 5V and 3.3V bus work with 3.3V bus pulled up to 5V and 3.3V bus.

avidtech, 2018-03-22:
Hmm, the best I could see on the SDA line was 3.3v, the pulses should be thin valleys the way I had it hooked up, as they were with my MS5611 I checked for reference since the scope wasn't getting anything but straight 3.3v DC with the BMP280. Too bad it wasn't talking or I'd be able to tell you what the dynamic range was. >:(

[5 more replies]