betaflight/betaflight #1583: BIG IDEA!!

Calrc, 2016-11-17:
I was pondering an idea.  On the receiver page you are able to see all your channels functioning within the GUI.  How hard would it be to use your transmitter via your FC in your craft as a game controller for flight simulators? Is this something that could be included in Betaflight?

mikeller, 2016-11-17:
You don't need to modify Betaflight for this. All you need to do is write a virtual HID driver for your operating system that polls the stick positions from the craft through MSP and uses them to emulate a game controller. Doing this is well outside the scope of the Betaflight project, but if you decide to give it a go yourself we'll be happy to assist.

jflyper, 2016-11-17:
@Calrc There are products you can connect your transmitter to your PC and make it look like a game controller, both wired and wireless. Many flight simulators already support this kind of device.

andre92, 2016-11-17:
I think he means for transmitters that cannot be connected to PC at all (no usb or anything). In that case it would indeed need to go via the flight controller using some driver that understands MSP and polls the FC for radio channels.

AndersHoglund, 2016-11-17:
http://www.simstick.co.uk/
https://hobbyking.com/en_us/wireless-9-in1-simulator-adapter.html
http://www.banggood.com/20-In-1-USB-Simulator-For-RC-G6_5-Phoenix-4_0-XTR-VRC2-FMS-p-89810.html
etc etc 
It is very easy to modify the last USB dongle for wireless, solder a piece of cable with a servo connector to GND, +5V and PPM inside the dongle and use your favourite PPM Rx. Been there, done that.
https://www.rcgroups.com/forums/showthread.php?t=1719953
https://www.rcgroups.com/forums/showthread.php?t=2039065

closed 2016-11-17 by AndersHoglund