betaflight/betaflight #1592: unreadable dump and diff output

TheAngularity, 2016-11-17:
@AndersHoglund 
your #1586 removed all standalone zeros from dump and diff output ... please fix quickly

diff output:

```
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# Betaflight/SPRACINGF3EVO 3.1.0 Nov 17 2016 / 22:17:06 (1dc9043)

# name
name HX200

# resources

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature -CURRENT_METER
feature -RSSI_ADC
feature -TRANSPONDER
feature LED_STRIP
feature AIRMODE

# beeper

# map
map TAER1234

# serial

# led
led 1 10,10::AW:
led 2 9,10::AW:
led 3 8,10::AW:
led 4 7,10::AW:
led 5 6,10::AW:
led 6 5,10::AW:
led 7 4,10::AW:
led 8 9,5::C:6
led 9 8,5::C:6
led 10 7,5::C:6
led 11 6,5::C:6
led 12 6,2::CW:
led 13 7,2::CW:
led 14 8,2::CW:
led 15 9,2::CW:

# color

# mode_color
mode_color 6 1 3

# aux
aux 1 12 1 1000 1200
aux 2 14 1 1000 1200
aux 3 14 1 1800 2000

# adjrange

# rxrange

# rxfail

# master
set min_check = 1000
set max_aux_channels = 2
set min_throttle = 1100
set max_throttle = 1950
set use_unsynced_pwm = ON
set motor_pwm_protocol = MULTISHOT
set motor_pwm_rate = 4000
set serialrx_provider = SUMD
set vbat_scale = 112
set align_board_yaw = 90
set gyro_notch1_hz = 
set gyro_notch2_hz = 
set acc_hardware = NONE
set pid_process_denom = 2
set blackbox_rate_denom = 4
set ledstrip_visual_beeper = ON

# profile
profile 

set dterm_notch_hz = 
set pid_at_min_throttle = OFF
set setpoint_relax_ratio = 40
rateprofile 

# rateprofile
rateprofile 

set rc_rate = 108
set rc_rate_yaw = 108
set roll_srate = 80
set pitch_srate = 80

```

AndersHoglund, 2016-11-17:
Ouch. On it.

AndersHoglund, 2016-11-17:
Reverted commit. Sorry for the mess...

TheAngularity, 2016-11-17:
thx 👍 ... closed now

closed 2016-11-17 by TheAngularity

mikeller, 2016-11-17:
> Reverted commit. Sorry for the mess...

Not your fault. The compiler warning is wrong, since `d /= base` can yield 0 for non-0 values of `d` with ints. I was going to fix the same spot as well...

AndersHoglund, 2016-11-18:
Well, cppcheck was correct. I just did a too Q&D (quick &dirty) fix. Could have spent some more time on it. The revert was also a bit of a panic to get some piece. Now I have a much cleaner and leaner version of those two conversion funcs. A lot of redundancy removed. They where really over worked and overcomplicated, fooled me.. 
Not sure I dare to PR the new version....  Well, OK. #1593.

mikeller, 2016-11-18:
I don't think it was, `d == 0` was possible without a division by zero being triggered in `dgt = num / d`, since `d /= base` in between can very well change d from non-zero to zero for int.

#1593 looks good.