betaflight/betaflight #1597: ESC telemetry implementation.

hydra, 2016-11-18:
ok, I might be missing something here, but `escTelemetryPort` is never used other than to hold a serial port reference, it's never used to read or write from.

That being the case I don't think the implementation of ESC telemetry follows the same pattern as other telemetry providers.

Telemetry providers in CF/BF collect information from the FC and SEND data via a serial port.

ESC telemetry is the reverse of that.  ESC telemetry COLLECTS telemetry data from an ESC for use by the FC.

Therefore, ESC should not be treated like other telemetry providers.

hydra, 2016-11-18:
@martinbudden @ledvinap @borisbstyle @basdelfos

Would love some feedback on this architectural issue.

hydra, 2016-11-18:
ok, so I did miss something, the use of the ISR for receiving telemetry :smile: `escTelemetryDataReceive`.

However, the architectural issue still stands.

The implementation of the ESC telemetry feature goes against the design of the 'telemetry providers'.

What you need is a 'telemetry collector' design.

I consider the current implementation a bit of a hack, issues are already coming up, such as how to configure the voltage/current meter sources, etc.  I feel that continuing to treat ESC Telemetry as a 'telemetry provider' instead of a 'telemetry collector' is a bad idea.

blckmn, 2016-11-18:
Thinking about this - the ESC is really acting as a sensor. The protocol is just serial. Perhaps the same pattern as per other sensors.

In theory we should really consider support for making the FC a master in a set of sensors. Supporting smart-port etc.

hydra, 2016-11-22:
yes, i came to similar concolusions, the escs are just sensors, they closest related sensor is the gps sensor and we certainly don't deal with the GPS sensor as a telemetry provider. 😄 

So, if we're in agreement here whats the next step to fixing this?

blckmn, 2016-11-22:
I'll raise a PR this weekend moving it around a little - for discussion - and it is very similar to GPS so I can see it being based on that.

martinbudden, 2016-11-22:
Yes, the ESC is acting as a sensor here. No need for a 'telemetry collector', since all sensors potentially act in that role.

@basdelfos , can you rename `esc_telemetry.c` to `esc_sensor.c` and move it to the `sensors` directory? And also get rid of the `escTelemetryPort` from your code?

basdelfos, 2016-11-22:
OK, I will fix this in the upcoming days.

blckmn, 2016-11-22:
Thats great @basdelfos

basdelfos, 2016-11-23:
@martinbudden, see https://github.com/betaflight/betaflight/pull/1646 for changes i made. Can you review this pr.

llambkin, 2016-11-28:
WOW! lots of stuff going on here. is it possible to use frsky hub telemetry stuff work if the FC is a master of a set of sensors? I have a frsky lipo monitor that pulls individual cells and spits out hub telemetry data. Its a bit of a waste of a uart tho just for that.....

mikeller, 2016-11-28:
@llambkin that's a good idea actually. It's not something that is currently supported, but I think it would make sense to add a 'FrSky Hub feedback' sensor source. Since FrSky Hub uses one way communication, it would then be possible to use one serial port for FrSky in / out, with Rx listening to sensor data, and tx sending the combined sensor data / firmware data to the RC Rx.

RBeaubien, 2016-11-29:
I have been following this thread with interest.  I am designing a new AIO FC and want to add ESC telemetry pads.  The KISS FC has these connected to UART3 TX. Is the consensus here is to do the same or have I missed something?  Just trying to get a head start on a potential future feature of ESCs.

mikeller, 2016-11-29:
@RBeaubien I think it would make more sense to connect them to the RX of an UART (for F3 it should be possible to use the TX, if the UART is set to bidirectional, but there is no benefit to this really).

RBeaubien, 2016-11-29:
I would agree, but it looked like an attempt was being made to duplicate the KISS method.  Wanted to avoid adding a "never implemented" feature if at all possible.  I still have a couple of weeks before I have to come to a final decision.  Hopefully a decision on the correct type of pin to connect to will be decided by that time. :-)

mikeller, 2016-11-29:
On second thought, I am not even sure if adding pads for ESC telemetry is really required at all. After all, users with KISS ESCs can just select the UART of their choice, and connect all ESC feedback lines to this UART's Rx pad. There is probably better use for board space than adding distinct pads for ESC feedback that all go to the same MPU pin (if nothing else, add pads for more MPU pins, allowing users to assign additional resources to them).

martinbudden, 2016-11-29:
> On second thought, I am not even sure if adding pads for ESC telemetry is really required at all. After all, users with KISS ESCs can just select the UART of their choice, and connect all ESC feedback lines to this UART's Rx pad.

I can see a convenience argument for ESC telemetry pads. It would make the build easier and neater, but as you say at a cost of build space. However, different people prefer different things, so it could be nice to have a board with this option in the marketplace.

RBeaubien, 2016-11-29:
I have plenty of space.  This is a AIO for an Alien frame.  If this is going to be a supported feature, I would like to add the pads to my board.  Just sorta hoping some decision is made before I need to commit to production.

andymac4182, 2016-11-29:
@RBeaubien Can't wait to see. Any idea how long till you have pics / specs up?

basdelfos, 2016-11-29:
If you have 'plenty' of space, than it would be the most convenient to add the pads for ESC telemetry near the signal and gnd pads for the ESC. Than also add separate pads for UART3 TX and RX. That way users who do not use ESC telemetry can use UART3 for other purpose. 
Maybe you can also take a look at the new LUX V2, it uses a bigger package of the STM32F3 with more resources and UARTs. That way you can assign ESC telemetry to a UART and still have plenty of available UARTs for SBUS/S.PORT and OSD or Logger.

RBeaubien, 2016-11-29:
@basdelfos  My plan was pads near the escs, and a jumper pad to give 2 options to connect (or not connect) them to the F4.

@andymac4182 It will be F4 CPU with ICM-20689 MCU, BlackBox RAM or SDCard, and OSD all connected via SPI.  Probably not out till January.

basdelfos, 2016-11-29:
@RBeaubien Nice configuration, will it be Alien only or standard 36x36 format?

RBeaubien, 2016-11-29:
Alien only.  PDB on that is structural.

basdelfos, 2016-11-29:
I know, that why I asked. That also confirms why you have plenty of space.

mikeller, 2016-11-29:
@RBeaubien: Now I can see your point about wanting the pads - didn't make much sense to me for a 36 squared board. :-)

hydra, 2016-11-29:
I think we can close this issue now, it's gone off-topic and the work has been done.

closed 2016-11-29 by hydra