betaflight/betaflight #1611: Receiver signal not being seen in Configurator in #729 for BeeBrain target

glenb2, 2016-11-19:


mikeller, 2016-11-19:
A bit more information would be helpful.

glenb2, 2016-11-20:
Steps:
Flash 729 to BeeBrain v.1d (Spektrum)
Go to Receiver tab in Configurator 1.8.4
ISSUE - Observe that all signals remain at defaults although the receiver is bound and controls are being moved
Flash 3.0.1 to BeeBrain v.1d (Spektrum)
Go to Receiver tab in Configurator 1.8.4
EXPECTED - Observe that all signals behave as expected when controls are moved

mikeller, 2016-11-20:
Stupid question, did you make sure that 'Serial RX' and 'Spektrum' are both selected in the configuration tab in configurator when running 3.1?

glenb2, 2016-11-20:
BeeBrain uses PPM, but I tried anyway and that doesn't seem to work either.  The exact same config on 3.0.1 is not working on 3.1.  Tried on 735 just now.

borisbstyle, 2016-12-22:
Reported as fixed in https://github.com/betaflight/betaflight/pull/1863

closed 2016-12-22 by borisbstyle