betaflight/betaflight #1612: Lost transmitter connection to board and beeper activates around 1180 on RMDO, Build 718/730

hippyengineer, 2016-11-20:
Build 718 and 731 BF 3.1 loaded to RMDO caused connection between Frsky XSR and RMDO to be lost. Testing DYS XS20A for Dshot (cap removal) and found that the beeper started a constant signal after 1180-1250 for all motors, whether on Dshot 150 (single motor esc cap removed and loaded A_H_70 16.43) or Oneshot 125 (no caps removed, A_H_70 16.45).

mikeller, 2016-11-20:
What protocol (SBUS / PPM) do you use for the connection between the Rx and the FC?

hippyengineer, 2016-11-20:
SBUS

borisbstyle, 2016-12-19:
Can you describe this better? So on oneshot and dshot there is no connection?

hippyengineer, 2016-12-19:
That is correct. I have reverted back to a 3.0 build now so I do not have the ability to test any further.

borisbstyle, 2016-12-20:
Can you also verify this with the latest build? Sounds honestly like ESC issue, but it is weird that it should work on 3.0.

closed 2017-08-17 by blckmn