betaflight/betaflight #1629: ACC X Y Z vibration measurment on telemetry output

wolkstein, 2016-11-21:
hello,
this is more a feature request than an issue.

i am a relative new betaflight user. i use for many years ardupilot in most of my copters. ardupilot has an nice feature to measure vibrations.  i personally never will missing this, after i noticed how useful it is. you can get an high pass  filtered output from your acc 's to measure vibrations from the system. the high pass filter remove all normal movements from the acc so that only system vibrations are left which will reported via mavlink "https://pixhawk.ethz.ch/mavlink/#VIBRATION" . this is a great feature to diagnostic your vehicle in flight to locate and optimise mechanical problems on your racer.  is there any chance to implement such a feature in betaflight. not implicitly mavlink based but in telemetry output. currently i noticed that you can get all acc values low passed filtered on the smartport telemetry system. fond in https://github.com/betaflight/betaflight/blob/master/src/main/telemetry/smartport.c#L702

it would be great if users have an option to select which kind of acc measurement will reported to telemetry.

i know that it is possible to diagnostic vibrations with the betaflight gui in motor config tab. but this is not the same if you get vibration diagnostic while you flight the copter. in this case you get relevant infos in which flight situation the system have mechanical problems. this is also more efficient than post flight diagnostic from logfiles. imo, with the direct feedback in flight you are able to provoke and locate the problems on your system very fast.  

/g
wolke

borisbstyle, 2016-12-20:
@wolkstein 
There is actually a blackbox feature what can record the vibration much faster than you could do via smartport or any other telemetry protocol. This means you can use higher sampling and see actually more information you would ever see on slow telemetry protocol.

Feel free to ask questions about Blackbox feature

closed 2016-12-20 by borisbstyle

wolkstein, 2016-12-28:
thx,
blackbox is great. the feature i request can not compete with black box diagnostic. it is more to get bad vibration notice(alarm) while flying. also it need no high telemetry refresh rate. if the averange from, for example, a period time from 250ms-1000ms of vibrations will measured than there is no problem to report this average value via telemetry. it helps in flight very simple to capture if something goes wrong on the system.
loose screws, dirty or unbalanced propellers or motors are detected immediately.
on taranis you simple use an script which give you an audio message alarm if an from user selected threshold for X Y Z vibrations are reached. then, for example, you simple hear "Warning, high XY Vibrations  (numeric value)" or "Warning, high Z vibrations" that is all.
Simple and save.

/g
wolke