betaflight/betaflight #1630: [Feature Request] Commence ESC beeps upon failsafe

braeden123, 2016-11-21:
It would be great to have a option that beeped the ESCs when a copter entered failsafe or had a mode in which ESC beeping could be activated by an Auxiliary channel. This would be beneficial for those who fly without a beeper and would work to alert location within a short distance.

I don't know if this is currently implemented or even feasible, but I figured its worth bringing it up.

flounderscore, 2016-11-21:
That would be a question for the ESC hardware developer (BLHeli).
Currently, the FC can't to anything about it.

I have the beeper timeout in blheli set to ~1 minute, i.e. after disarming
the escs start to beep after 1 minute. If you're stuck waiting on the
starting line for a race, I just do a quick arm/disarm to reset the timer.

martinbudden, 2016-11-21:
Actually, this is probably possible with DShot, since that protocol has a facility for sending commands to the ESC. @blckmn - I'm not familiar with the DShot commands, is beep among them?

lichtl, 2016-11-21:
Yes. Should be possible with Dshot (KissFC / KissESC) do already support this. Not sure about the implementation in BLHeli and the other Dshot ESC's

Of cause it require some checks to not "beep" while flying.

flounderscore, 2016-11-21:
Someone once proposed to use a low us window for analog escs, e.g. 800-900 us = beep, but again, someone would have to implement that in blheli.

blckmn, 2016-11-21:
Definitely with DSHOT we can disarm the motor, and then start playing a variety of tones! :)

varkokonyi, 2016-12-10:
It is possible to not output a valid signal to the ESCs, then they beep if I am right

mikeller, 2016-12-11:
@varkokonyi That's already a thing, albeit it activates after a set periond of inactivity (default 10 mins for blheli). No need to reinvent the wheel, and probably a safer bet to use inactivity as the trigger, for all we know the Rx could have been disconnected during the crash...

borisbstyle, 2016-12-20:
I will close this "feature request" as the limitation in this one is the current ESC protocol.

In the near future this will be possible with DSHOT and it will be implemented in the firmware.

closed 2016-12-20 by borisbstyle