betaflight/betaflight #1631: disarm motors when lipo hits low voltage

trojanGoat, 2016-11-21:
Hi Betaflight team,

I know there's a low voltage cut off that lands the copter,  but I'm not exactly sure how that works.  It seems like I'm still killing my batteries.  This is more of a problem on the qx90 micro when I dont' have a beeper or osd.  Is there a way to get the motors to completely disarm when the voltage is at a certain level?  

Maybe I just need to raise my low voltage cut off?

mikeller, 2016-11-21:
Probably using a timer on your transmitter (or on your phone or a kitchen timer ;-)) is the best way to avoid killing batteries.

Depending on the Rx you got on your QX 90, you could also try this in order to get telemetry feedback of the battery voltage to your Tx: 

https://sites.google.com/a/042.ch/mikesthings/home/getting-telemetry-on-an-eachine-qx-95

trojanGoat, 2016-11-21:
That's a great idea,  I've setup timers before, that is only good if you fly at a consistent rate.  I don't have telemetry unfortunately.  My only other idea is to solder an led light to the buzzer feed.  Either way I'm just looking for a way to prevent my lipos from prematurely dying off.

Jahnkeanater, 2016-11-22:
Maybe disabling the motors on brushed targets isn't a bad idea. Or set the brushed motor output to check if Vbat is critical.

I think most people prefer their quads not to fall out of the air randomly.

Szpero, 2016-11-22:
Timer is for most cases enough:
- 8520 motors + 55mm ladybird. I'm using simple rule. Ah x 10 - 1 = minutes. So for 0.6Ah I'm flying for 5minutes. Its fits well within 500-750mah lipo range.
- 1020 motors + 65mm gemfan. Rule same but -2 minutes. Those ones are really lipo killers. Even graphene 1s starting to be warm.

Telemetry: working well and go along with tutorial from link above.

One more improvement for RX worth to be mentioned: get rid of wire antenna and put there dipole (antenna with barrel).

2black0, 2016-11-28:
1. timer
2. volt telemetry
3. osd

mikeller, 2016-11-28:
Closing this, solutions outlined in discussion.

closed 2016-11-28 by mikeller