betaflight/betaflight #1632: IBUS Telemetry

Jahnkeanater, 2016-11-22:
Any chance that IBUS telemetry will be implemented in the next stable build? Looks like CleanFlight has it working. :)

blckmn, 2016-11-22:
I'll have a look into this one.

andymac4182, 2016-11-22:
I think this is a duplicate of https://github.com/betaflight/betaflight/issues/1035

Jahnkeanater, 2016-11-23:
That's from 4 months ago though. Hydra finally released IBUS telemetry in the stable build 12 days ago. I don't think it would be too hard to get it working now.

andymac4182, 2016-11-23:
@Jahnkeanater If you read the latest comments someone has got it working in Betaflight it just needs to be pulled in.

LightningStalker, 2016-11-28:
This would be nice to have in my Turdigy i10

borisbstyle, 2016-12-19:
https://github.com/betaflight/betaflight/issues/1035

borisbstyle, 2016-12-19:
please always check if an issue already exists

closed 2016-12-19 by borisbstyle