betaflight/betaflight #1633: AIR32 (Lemon) voltage scaller not working

Szpero, 2016-11-22:
Voltage source connected to L/G pins. Voltage reading are working - value depends on input voltage value.
Problem is with scaller itself so user cant adjust readings to correct values.
Changing even from 0 to 255 doesnt affect on final shown value.

mikeller, 2016-11-22:
Where do you read the final shown value from? If it's from the configurator 'configuration' tab, are you aware that you have to hit 'Save & Reboot' for the changed scaling to take effect?

Szpero, 2016-11-22:
Configurator software connected by USB.
If I remember well it was something about 25.5v. Swapping cables between amount of cells had influence on value.
On every change of scaller I saved it and FC was rebooted. I didn't noticed change on small values so I tried 0 and 255 value without effect.

mikeller, 2016-11-22:
Ok. A reading of 25.5V (255 * 0.1V) means that your voltage scaling is so far off that you are over the limit for this display to work. Also, a scaling of '0' means the battery voltage input is not used for voltage calculation at all. I would suggest that you test again with values of 1 and 10 for the scaling. If these both come out as 25.5V then the battery voltage input on your board is broken.

Szpero, 2016-11-22:
Closing case.

Different PC, fresh install of bF and L/G connected to lab power supply.
Don't know why but now scaller is working and all is trimmed to correct values.

Thanks for support.

closed 2016-11-22 by Szpero