betaflight/betaflight #1635: Smartport Telemetry still not working on F4 targets?

HaseebNain, 2016-11-22:
I have been working on getting smartport working on F4 targets. 

Because the F4 does not natively support UART inverting, I used the taulabs inverting circuit on UART3 in an attempt to get some sort of S.PORT connection. 

[image]

I have attempted to turn off inversion "telemetryConfig->telemetry_inversion" in Smartport.c and used the inversion circuit instead. This did not seem to work with the REVO (making sure to point the Firmware to use UART3 for Smartport telemetry). 

After trying the inversion circuit I went ahead and followed @blckmn's Smartport tutorial  [Smartport, the FrSky X4RS, and Betaflight](https://blck.mn/2016/06/smartport-the-frsky-x4rs-and-betaflight/). This method worked and made me wonder why the inversion circuit itself wasn't working. 

Is there possibly an issue with the F4 and the original porting of the F3 code that may have been overlooked? Perhaps still using a pseudo one wire smartport and not accepting the bidirectional communication through the RX?

Are there any F4 boards that have working SmartPort telemetry? I know that the Revolt is saying they will be able to support SmartPort through software in RF1 but are unable to support it at the moment.

HaseebNain, 2016-11-22:
@blckmn Were you able to use a similar inversion circuit to get your BlueJayF4 working with Smartport?

With this circuit I was able to use an arduino to push a PWM signal and saw it inverted through an oscilliscope but was never able to get it to establish SBUS without using your workaround.

HaseebNain, 2016-11-22:
Could the issue be related to #1386 and #1499 as UART (msp) information seems to be inconsistently corrupted as shown by @jflyper?

F4 msp TX and RX
[image]

compared to the F3 msp TX and RX
[image]

blckmn, 2016-11-22:
@HaseebNain it is possible the MSP corruption is similar issue for smartport.

However I would check that the bi-dir option is not on for smartport in F4 - otherwise the RX pin is effectively muted, and everything goes through the TX (hence why the 1 wire option works).

Hazakins, 2016-11-22:
@blckmn would that be in the UART driver code?
Thanks

HaseebNain, 2016-11-22:
Smartport by default has BIDIR enabled.

In smartport.c ("src/main/telemetry/telemetry.c" line 334): 

    portOptions = SERIAL_BIDIR;

I changed this to

    portOptions = SERIAL_NOT_INVERTED;

and also set

    telemetryConfig->telemetry_inversion = 0;

in config.c ("src/main/fc/config.c" line 369)
to account for the hardware inversion. Still no signs of Smartport communication with the serial set normally. 

Is there something else with the UART that would mute the RX? It might be functioning properly just missing the initial RX/TX Smartport "handshake"?

HaseebNain, 2016-11-23:
Cleanflight it seems was able to get smartport working with the dual invertor circuit. 

https://github.com/cleanflight/cleanflight/issues/818

@netsplits I see that you were able to get smartport working with Cleanflight using a dual invertor inversion circuit. Is there anything with the Betaflight smartport code that seems to be out of order when compared to the firmware you worked with? The firmware seems incredibly close.

I noticed that Betaflight sends information when not idle||start_stop
https://github.com/betaflight/betaflight/blob/master/src/main/telemetry/smartport.c#L269
    
    else {
        serialWrite(smartPortSerialPort, c);
    }

whereas in Cleanflight data is sent continously
https://github.com/cleanflight/cleanflight/blob/master/src/main/telemetry/smartport.c#L171

    serialWrite(smartPortSerialPort, c);

Perhaps Betaflight is sending information during the timeframe radio is attempting to send information to the FC?

I believe that smartport inversion is incredibly close to a working state.

slawek89, 2016-11-25:
Can anyone tell me how it is with SmartPort?
Single connection work with F4 targets (simply connect to the UART TX and inversion on)?
Is it something else, when we combine the input and output (any additional options?).
It is necessary to add hardware inversion TX and RX or it can be done via software?
PS. Sorry for dumb questions but i'm not good with software magic.

jflyper, 2016-11-25:
@slawek89 That sounds like a support question. Please come to the RCGroups' thread. Which board are you using?

slawek89, 2016-11-26:
Actually I design my own F4 board for few friends. So I would like to know what hardware place on it. On other F3 boards SP works fine to me. 
I want to use only frsky xsr and I don't know if I have to place only inverter on TX or circuit like in first post.

jflyper, 2016-11-26:
@slawek89 You have to consider a couple of factors:

1. S.PORT is a bi-directional protocol.
2. Half-Duplex mode turned on by S.PORT support code connects RX to TX internally, and RX pin will be ignored.

Above collectively suggests that if you are going inverter way, the inverter must be bi-directional, and requires some non-trivial circuit similar to the one shown in the first post (I say similar because the circuit doesn't consider ignorance of RX pin in HDX mode). I suggest that you go non-inverting hack on XSR.

jflyper, 2016-11-26:
@HaseebNain 
>  noticed that Betaflight sends information when not idle||start_stop

No, that is not **IDLE**, but **DLE** (Data Link Escape). The lines you are looking at is trying to encode data bytes that has special meaning in data framing control so they are not mis-read as framing control at the receiving end (e.g., What if you want to send a byte that represents end of the frame as a part of the data, etc.)

Transmission timing is controlled by the receiver (or the master on the bus polling sensors), so the S.PORT support code can use the bus whenever it is requested for the data.

I think there are several things you have to look at elsewhere, including how S.PORT code handles  data received or the reception event itself while it is sending; things sent will be received as they are sent with your circuit and full duplex mode.

RipperDrone, 2016-11-27:
Pls let a noob understand what you are going for, @HaseebNain:

There is no way around the need for a hardware telemetry signal inversion if we want to use it on an F4 FC (like Omnibus/Flip) UART3 port, since only UART1 is inverted and used for S-Bus / receiver.

We can do an S.Port receiver hack (described in rcgroups) to get it to work.

The purpose of what you are trying here is to implement a hardware circuit on your custom F4 FC version which will then eliminate the need to do the hard mod hack on the receiver's S.Port line

All correct? Or are you looking for a way to software mod telemetry input into a native F4 FC? (which until now I thought is impossible...)

slawek89, 2016-11-27:
@jflyper 
Thanks for explanation. One question. SmartPort won't work on F4 targets like in F3?
So there is only one way to work SP (reciever hack)?

jflyper, 2016-11-27:
@RipperDrone All correct.

@slawek89 F3 has internal inverters on all USART channels. F4 has none thus requires external inverter wherever the signal is inverted.

Why F3 has them and F4 doesn't? Because F3 is newer than F4, and someone at ST micro found out it would be useful to have them when they started to design F3.

slawek89, 2016-11-27:
@jflyper Just out of curiosity, it is possible to make software inversion on F4 or it is impossible?
Sorry if it isn't question for this place.

HaseebNain, 2016-11-27:
@ripperdrone Correct, the hardware circuit has been implemented and tested to work. While the hardware hack works, it isn't ideal. 

I'd like to see if software inversion is possible, which might be easier said than done. Or at the least just get the hardware circuit + firmware working so the community can simply use any F4 board and enjoy S.Port without having to go through the hack.

Faduf, 2016-11-27:
You can maybe use softserial, like it is done on many F1 boards.

jflyper, 2016-11-28:
Softserial may work, but it's a cycle hog.

jflyper, 2016-11-28:
Is there any SmartPort master emulator that I can use to debug SmartPort slave code? Something that runs on Arduino, may be?

slawek89, 2016-11-28:
Maybe you can use this:
http://www.rcgroups.com/forums/showthread.php?t=2245978

jflyper, 2016-11-28:
@slawek89 This is good. Will try it. Thanks!

Linjieqiang, 2016-11-28:
[image]
Was this circuit correct?

jflyper, 2016-11-28:
@Linjieqiang I believe it does what it was designed to do, but currently has some trouble interfacing to smartport code in BF.

Linjieqiang, 2016-11-28:
@jflyper So I want use this cricuit in my fc board to support S.port.The BF can support it later.

jflyper, 2016-11-28:
@Linjieqiang At your own risk. And also note that you would need some additional arrangement unless you designate a UART specifically for S.PORT.

Linjieqiang, 2016-11-28:
Yes.I designate the UART3 specifically for S.PORT.Is it ok?

HaseebNain, 2016-11-28:
@jflyper Yea that would be fine, I just started looking over that RCGroups thread too. Let me know whatever and whenever you want to debug anything!

Hazakins, 2016-11-28:
@jflyper I could possible help too. I have a XSR and a Revo f4 with this circuit wired up.

codygarvin, 2016-11-29:
I am working with @HaseebNain on trying to get this circuit to work. My only questions is that due to the wiring of this circuit anytime the MCU is transmitting down UART TX it's going to be reading a duplicate of the data being transmitted back into UART RX. Does the code already ignore the RX buffer during TX writes? Since the drive side of the TX inverter is on the gate side of the RX inverter you will see RX data anytime you TX out.

jflyper, 2016-11-29:
@codygarvin That's what I was wondering for few days (see [comment ](https://github.com/betaflight/betaflight/issues/1635#issuecomment-263064157)).

You can embed some assignments into debug[] variables to see how states are transitioning. Can you do this?

HaseebNain, 2016-11-29:
We haven't used debug[] before but can look into enabling and using it to see the state changes.

codygarvin, 2016-11-29:
@jflyper So does the SPORT receiver ask for telemetry data and the MCU processes the request and responds? Or does the MCU just feed out the data with appropriate headers and the receiver just replies with handshakes? Either way, the code will definitely have to ignore or block bytes on the RX line when using TX. Even if you went with a 2-channel XOR implementation for inversion you run into the same issue.

jflyper, 2016-11-29:
It's a request (receiver) - response (fc) protocol.

First few steps of `debug[]` debugging would be:
- Call count for `smartPortDataReceive`.
- Count `smartPortDataReceive` flagging `smartPortHasRequest`.
- Call count for `smartPortSendPackage`.

jflyper, 2016-11-29:
@Linjieqiang 
> Yes.I designate the UART3 specifically for S.PORT.Is it ok?
It's your board and your decision to dedicate some resource for some specific purpose...

jflyper, 2016-11-29:
@codygarvin And don't forget to turn off the `SERIAL_BIDIR`.

You folks don't have logic analyzers?

Linjieqiang, 2016-11-29:
How about using softSerial to implement the S.port?@jflyper

jflyper, 2016-11-29:
@Linjieqiang Read this [comment](https://github.com/betaflight/betaflight/issues/1635#issuecomment-263158760).

SoftSerial is my last option.

Linjieqiang, 2016-11-29:
Oh forgive me for not seeing clearly. But for F4 board ,SoftSerial for S.port is a nice solution.

Hazakins, 2016-11-29:
@Linjieqiang Softserial would be nice as a backup, but like @jflyper mentioned we dont want to use the MCU overhead.

Linjieqiang, 2016-11-29:
@Hazakins So can you come up with the perfect solution?

Hazakins, 2016-11-29:
no, i cant but these guys are working on it. If your dead set on softserial you may want to open a feature request and start working on it there.

Linjieqiang, 2016-11-29:
Absolutely,I want to use hardware usart to solve the S.port but if has no idea I only choose softserial to solve it.

HaseebNain, 2016-11-29:
@jflyper Yea, all tests I've run have been with BIDIR set to SERIAL_NOT_INVERTED.

Unfortunately we don't have a Logic analyzer, so for debugging purposes I've been using the status LEDs on the boards. We also have an oscilloscope that we can use to count and check the general state of bits and information being transferred.

Looking into getting a Logic Analyzer, any recommendations?

AndersHoglund, 2016-11-29:
https://www.saleae.com/

jflyper, 2016-11-30:
@HaseebNain 
> Yea, all tests I've run have been with BIDIR set to SERIAL_NOT_INVERTED.

I'm confused.
[https://github.com/betaflight/betaflight/blob/master/src/main/telemetry/smartport.c#L323-L332](https://github.com/betaflight/betaflight/blob/master/src/main/telemetry/smartport.c#L323-L332)
```
void configureSmartPortTelemetryPort(void)
{
    portOptions_t portOptions;

    if (!portConfig) {
        return;
    }

    portOptions = SERIAL_BIDIR;
```
Here, the assignment to portOptions should be 0  (or `SERIAL_UNIDIR`).
Inversion setting (`SERIAL_NOT_INVERTED` and `SERIAL_INVERTED`) is don't care.
So, make sure portOptions is zero when calling `openSerialPort`.

> Looking into getting a Logic Analyzer, any recommendations?

I would recommend the Saleae, too. I myself use Logic 8. It will be a matter of minutes if not seconds to solve this kind of problem with one of these.

HaseebNain, 2016-11-30:
@jflyper Are you located in the US? Maybe we could let you borrow our XSR and Taranis for debugging purposes?

jflyper, 2016-12-01:
@HaseebNain Unfortunately, I'm not in US (or anywhere near)...

blckmn, 2016-12-01:
Once I am back home I can take a look at this.

I have an inverter circuit on the BlueJay so it can be used for testing.

HaseebNain, 2016-12-02:
@jflyper Dang, no worries. I'll keep trying the things you mention. I did try with portOptions set to UNIDIR as well when I was working with the NON_INVERTED setting. The options still seemed not equal to zero, this was checked using that LED enable function and a delay (in case the LEDs are set off to quickly).

@blckmn That would be great! Having a board with the invertor circuit already embedded will definitely help in the debugging process.

blckmn, 2016-12-11:
https://github.com/betaflight/betaflight/pull/1803

allows for turning off the bi dir option for use with the circuit shown.

jflyper, 2016-12-12:
Any progress yet?

BTW, soft serial now works with inverted S.PORT (iNav).

HaseebNain, 2016-12-12:
@blckmn just tried #1803 on an F4 with the Inversion circuit, nothing seemed to show up on the Taranis. Were you able to test this on your end, how did the BlueJay work with new PR?

Simply defining the target as an F1/F4 should be enough to dictate whether that PR is in effect, otherwise I think for long-term having them a part of the serial port configuration would be for the best.

@jflyper I just saw the iNav merge, they mentioned a slightly different inversion circuit. The values of the resistors and type of mosfet differs but should be equivalent to the tauLabs circuit.

https://github.com/iNavFlight/inav/commit/32368b7db63a0141eeb4771a0142b7e4d0250a18

[image]

blckmn, 2016-12-17:
[image]

Is the circuit used on a small inversion circuit I have been testing with. On the BlueJayF4 (rev 3, it is effectively the same, but the diode needs to be added - just between the TX and RX pins - as the XOR pulls low).

This is working fine for me on the BlueJayF4 with SERIAL_BIDIR turned off.

Linjieqiang, 2016-12-17:
@blckmn Is this schematic for Smartport working well?

blckmn, 2016-12-17:
It is working for me :)

blckmn, 2016-12-17:
[image]

After installing the diode on a Rev3 (and using the updated https://github.com/betaflight/betaflight/pull/1803) smart port is working.

borisbstyle, 2016-12-19:
closing this one as it is hardware limitation on F4's

closed 2016-12-19 by borisbstyle

HaseebNain, 2017-01-05:
There is an issue with the taulabs inverting circuit.

The incredibly large output resistor (39k) is too large and depending on the MOSFET used it can limit the current draw and prevent fast enough switching. This can be remedied by reducing the baud rate of the smartport UART.

@blckmn and @borisbstyle  With how baud rates are stored within betaflight, is it possible to introduce smaller baud rates to help with this issue as more F4 boards begin introducing the smartport hardware to their setups?

blckmn, 2017-01-05:
@haseebnain the sport baud rate is fixed at 115200 or 57600 (can't remember which). The protocol is dictated by the RX, not the FC. The RX polls sensors - asking each sensor in turn to send its data back (the FC is just a series of pseudo sensors). The idea being that one wire can chain sensors together. 

So given the RX is sending the first transmission in this bidirectional protocol - I can't see how lowering baud rate will actually work. Have you tested lower baud rates?

HaseebNain, 2017-01-05:
We were testing the MOSFET itself and were able to get the smart port working through the example rcgroup thread: http://www.rcgroups.com/forums/showthread.php?t=2245978.

We used that example to spoof airspeed and pushed it through the taulabs inverter circuit. We had to change the drain pull down resistor on the PMOS from 39k to 10k and were able to use the 57600 baudrate. It is possible that the 39k resistor could work with a low power MOSFET that does not need as much drain current. Most common MOSFETs will require more current than a 39k can provide on 3.3 V.
 
After toying with an X4R we found that it uses a fixed baud rate of 57600.

[1 more reply]