betaflight/betaflight #1650: LED Strip not working right on Omnibus F3 Pro with SK6812 LEDs

Zethyl, 2016-11-24:
Issues with LED Strip on OMNIBUS F3 Pro target OMNIBUS 3.0.1

board has a dedicated LED Strip connector

15 LEDs total. #0-8 are a strip connected to the strip of #9-#14 by thin servo wire

SK6812 RGBW LEDs

-no control of led #12+, always white
-random control of #0-11, if i set all to red i get R,G,B,R,G,B,R,G,B,R,G,B,W,W,W if i set to white i get all white, if set to blue i get B,R,G,B,R,G,B,R,G,B,R,G,W,W,W

AndersHoglund, 2016-11-24:
I assume you have read the documentation on LED-strips, if not do that: https://github.com/cleanflight/cleanflight/blob/master/docs/LedStrip.md
I also assume you have tried the required series diode modification, described in doc above:

> If you have LEDs that are intermittent, flicker or show the wrong colors then drop the VIN to less than 4.7v, e.g. by using an inline diode on the VIN to the LED strip.

If all that is done and you still have problem, there might be a compatibility problem between the supported WS2811/WS2612 LED types and your SK6812. There are slight differences in timing specifications.

Zethyl, 2016-11-24:
Yes, I have read the documentation. I have also installed a diode. Still having issues.

Is this a problem with the timing or due to the fact that these are RGBW?

AndersHoglund, 2016-11-24:
Do not use USB to power the LEDs. Use a separate +5V regulator from flight pack. LED-strips usually draw too much power. 
I have only tested WS2812B and I have not measure the signal timing, so I really do not know if SK6812 are fully compatible. Compare the specs, and get some WS2812 and try them. Dirt cheap on BG. http://www.banggood.com/5Pcs-CJMCU-8-Bit-WS2812-5050-RGB-LED-Driver-Development-Board-p-961344.html

AndersHoglund, 2016-11-24:
Interesting co-incidence. Got my 5 pcs of WS2812  from BG today. And they work just fine without the series diode mod. Seems to be a difference where the Adafruit Neopixel WS2812B is more sensitive than those cheap WS2812 (non-B).

Zethyl, 2016-11-25:
The diode did not change the behavior with the SK6812 either.

I just came to the realization that the distributor I purchased this strip from may not exactly know what they are selling but here is a link to what I purchased. [](http://a.co/eZ4K310) They appear to be 32-bit RGBW LEDs like this [p2757_SK6812RGBW_REV01.pdf](https://github.com/betaflight/betaflight/files/612494/p2757_SK6812RGBW_REV01.pdf)

Other than the 24-bit / 32-bit difference there appears to be a huge difference between the PWM frequencies. I should also add that the logic input voltage is specified to be +/- 0.5V of the supply voltage, that means I am taking the diode out.

I found an issue in cleanflight that could be related... [](https://github.com/cleanflight/cleanflight/pull/2323)

AndersHoglund, 2016-11-25:
OK, 6812 is incompatible and not supported. Closing.

closed 2016-11-25 by AndersHoglund

gaelj, 2016-11-26:
pitty the branch wasn't merged at the time... i'll rebase it some time

brandonhill, 2017-10-09:
FWIW, I'm having this issue with WS2811 LEDs with diode (~4.3Vin) on an Omnibus F3 mini.