betaflight/betaflight #1653: Add battery capacity warning settings

nhadrian, 2016-11-24:
Hi,

I'd like to ask an option to set users warning limit for battery capacity readings. For example add a **battery_capacity_warning_limit** parameter between 0 ... 100 values (ie. 10 by default) that represents remaining capacity warning limit setting. Also please include this type of warning for different telemetry protocolls.

The reason is that using battery voltage for alarm is not too accurate especially when flying hard 3D in cold weather because you'll get many false alarms and also the cell voltage will drop down quite fast at the end of capacity, maybe pilot will not have enough time to land safety without over discharging batteries.
Adding timer in transmitter menu also not a good option because different tracks needs different flight style and current consumption. 

But with a properly calibrated current sensor mAh reading is also quite accurate, so setting up a percentage limit makes sense in my oppinion.

Thanks in advance.

Regards
Adrian

mikeller, 2016-11-25:
Hi Adrian.

Couple of points:
- the warning you are proposing will obviously be dependent on the battery capacity that is configurred in the firmware, and hence not work properly for users that use batteries of different capacity on the same craft;
- the mAhDrawn / remaining battery capacity is already sent via telemetry feedback;
- as a conclusion, it is probably easiest (and already possible) to set up the battery capacity warning on the Tx, where battery capacity can be configured in an easier way than in the firmware;
- with respect to the battery voltage warning coming on on hard usage, you should try playing around with `vbat_hysteresis`. Setting this will introduce a voltage hysteresis level that allows the voltage to drop below the configured minimum voltage before an alarm comes on, but keeps the alarm on until the voltage goes above the minimum voltage again;
- nonwithstanding this, I can see the benefits of your suggestion, i.e. to add a consumption warning that is communicated by beeper / LED strip, without reliance on other infrastructure.

My proposal would be to add configuration settings for `vbat_alarms` (on/off) and `consumption_warnings` (on/off), with `vbat_alarms` switching the existing alarms on / off, and `consumption_warnings` switching on a warning (slow beeps) when 10% of the configured capacity is remaining, and an alarm (fast beeps) when 0% are remaining.

Your thoughts, @nhadrian?

nhadrian, 2016-11-25:
Hi, thanks for your feedback. Of course this type of warning needs properly configured current monitoring and battery size. But since the battery size is 0 by default, that means once the battery size is configured by user in cli, user knows what is doing, so this type of warning will be relevant. If the default warning level will be 0%, there is probably no need for extra parameters switching different alarms on/off. If percentage is 0 then consumption warning is off, else on.

Related different battery sizes on a craft, battery size cozld be stored along with PID profiles, since  different battery sizes probably needs different PID settings, so switching profile also could handle the battery sizes. 

Related tx warning, not any system handles warning limits. Ie. I'm using HOTT, and there are not any  options for warning level settings in tx menu, it simply uses the FC warnings instead.

Also a visual and audio feedback is needed for configurations without telemetry.

Your suggestion sounds great, but we should let the user setup the remaining percentage because different situations needs different remaining battery (I mean for example on a short track 10% is enough but on a long track maybe 20% needed for safety return and landing).

mikeller, 2016-11-27:
Implemented by #1658.

closed 2016-11-27 by mikeller