betaflight/betaflight #1664: Blackbox issue with 8k/8k settings?

limhyon, 2016-11-25:
Hi,
I had a trouble of unknown twitches, but the twitch was gone when I turned off the blackbox.
I reckon that the problem was because of blackbox, it was trying to store data in 8k.
Was there any issue related to this?

jflyper, 2016-11-25:
@limhyon Can you supply us with at least:
(1) Your board
(2) Your storage type

jflyper, 2016-11-26:
P.S.
If you are recording at 1:1 on 8k, it's just a waste of everything. 2k effective is enough, 1k is fine for most cases.

borisbstyle, 2016-12-19:
@limhyon maybe your CPU load was too high on 8k/8k

closed 2016-12-19 by borisbstyle

hyonlim, 2017-01-17:
Hi,
I am  using custom made, F4 board.
CPU load is pretty low when 8k/8k setting.
I've tried 1k blackbox setting, still inconsistent cycle time.
What about another board? have you tried to log blackbox with 8k/8k  setup?