betaflight/betaflight #1675: problems running gdb on F3 target

fishpepper, 2016-11-26:
I have two problems using gdb on F3 targets:

1) first, i have to disable VCP, otherwise i get stuck in an endless usb irq etc loop - any hints?

2) second, i have to comment out the line 196 of adc_stm32f30x.c:
`while (ADC_GetCalibrationStatus(adc.ADCx) != RESET);`
Otherwise i get stuck at this line. 

Has anyone experienced the same? Could this be caused by not having the reset line connected to my stlink?
I only connected CLK, DATA and GND.

hydra, 2016-11-29:
your oscillator might be dead/dying on the FC.

fishpepper, 2016-11-29:
The strange thing is that when i do not run gdb everything works as expected. VCP works, the target starts up and the fc is working fine. Can someone confirm that a gdb session is working fine with a vcp enabled target?

AndersHoglund, 2016-11-29:
I just had an Eclipse/GDB/Segger/J-Link session with a F4 Revo clone. Found the VCP MSP jumbo frame bug that way, #1683 . Power from VCP USB, J-link USB adapter only has Clk, IO and gnd connected to SWD port. Mine don't even have a reset line... 
Can't recall I did anything special when installing the Eclipse plugins, but it was quite some time ago.

hydra, 2016-11-29:
yeah, all works fine here @fishpepper

MJ666, 2016-11-29:
Im also using Eclipse Neon, STLinkV2 (Discovery Board F3/F4), OOCD and GDB on Windows 10 for debugging. The AlienFlight F3 and F4 both are VCP targets and I debug the code in my development environment without problems. For the AlienFlight F3 I use the Discovery F3 board itself and for the AlienFlight F4 I have an Core405 board with sensors and some additonal hardware connected to debug and "simulate" the AlienFlight hardware. The real AlienFlight boards don't have SWD pins exposed.

fishpepper, 2016-11-29:
thanks for your posts. i got it working.... I was using st-util as a gdb server the whole time. Seems like this does not work reliably. I switched to openocd and not i can debug without any issues! 
I found it very strange that the adc code was getting stuck, probably some registers are not initialised/cleared with the st-util server :-\

closed 2016-11-29 by fishpepper