betaflight/betaflight #1676: The 3d_deadband_throttle is changed from 50 (default) to 1000

MJ666, 2016-11-26:
This will always happen if the BF configurators configuration page is saved even if there is noting changed. not sure if this is related to the firmware or the configurator itself. Using 3.1 build from today with configurator 1.8.4. Could not really find anything in the firmware wich could cause this.

MJ666, 2016-11-26:
I can only see the initialization in config.c and writing the parameter via MSP. This are the only two locations where it is written to this parameter directly from the firmware code. Likely the value will be changed from the configurator?

mikeller, 2016-11-27:
I think it is an off by one error in the MSP message format, so it's inbetween the two. ;-) But to my recollection we had this exact bug fixed in the past, and this is probably a regression.

rotcehdnih, 2016-11-27:
Watch out MJ666 3D mode is very broken in 3.1 after some mixer changes  , Please watch this video to see the current issues that are being worked on .. 

https://www.youtube.com/watch?v=U42MLzdquhI

MJ666, 2016-11-29:
The 3d_deadband_throttle problem is solved by #1694. Thanks to @mikeller .

closed 2016-11-29 by MJ666