betaflight/betaflight #1679: 3.0.1 pikoblx issue

jeffwhu, 2016-11-27:
Just got kombini FC and flashed it to pikoblx 3.0.1, ,my quad I put it on was uncontrollably yawing, checked all settings in betaflight and taranis everything good.
Ref laser the FC to pikoblx 3.0.0 and the problem went and flies good now, not sure why 3.0.1 wouldn't work.
Just an advisory or has anybody else had the same issue?

Jeff

jflyper, 2016-11-27:
@NightHawk32 Is PIKOBLK target properly maintained?

jeffwhu, 2016-11-27:
Don't know what you mean by"target properly maintained"?, sorry

NightHawk32, 2016-11-27:
@jflyper I think it is. I am not aware of any big issues about this target.
Just took one of my Kombinis and flashed 3.0.1 again to see if I can reproduce this issue. Everything behaves as expected.
@jeffwhu Can you check if the yaw issue is caused by strange sensor values or by issues regarding rx input?

jeffwhu, 2016-11-27:
Had a look at sensors all appear fine, rx input correct.
Flashed 3.0.0 to kombini and works fine.

closed 2016-11-28 by jeffwhu