betaflight/betaflight #1681: For disscussion - RTOS for betaflight

martinbudden, 2016-11-27:
This issue is to discuss the use of an RTOS with betaflight. Pros/cons, which RTOS to use etc.

jflyper, 2016-11-27:
Additional items to consider:
- Migration strategy; disruptive, non-disruptive, etc.
- Bus resource scheduling?

ledvinap, 2016-11-27:
(copy from related threads)
With RTOS, current system may run as single thread - scheduler and all. This will allow to use current simple model that ignores proper synchronization.
Parts of code may be moved to separate threads eventually:
- interrupt handlers may be simplified/partly moved to task.
- bus arbitration
- gyro + PID + motor control - proper synchronization of control inputs (RC, attitude, ... ) will be necessary, but that shouldn't be that hard

ledvinap, 2016-11-27:
Ad bus resource - One possibly quite easy way is to run separate thread for each bus, calling driver-supplied functions.
The thread will decide which driver shall get the bus (priority/round-robin based) and call drivers function.
Bus request called from this function will simply preempt until data are ready, so implementation will be almost identical to current code with almost free background processing.
The function shall do necessary work and pass data to current scheduler thread (using proper synchronization). Driver line baro will be queued multiple times per read, waiting for results outside of driver function.

IMO this should blend quite easily with current code, allow complicated device drivers without complicating bus driver and shall be reasonably efficient.

blckmn, 2016-11-27:
I quite like this incremental approach proposed by @ledvinap Having each bus in a thread will require some effort in the elimination of the while loops (the request - wait - receive) releasing cpu whilst in the wait state. Device (or DMA) ISRs can then be used on the receive and to synchronise.

ledvinap, 2016-11-27:
@blckmn : Not sure what 'while loop you mean;. Current code code accesses bus only when actively using it (busy-waiting for result), but lets other tasks run when waiting for processing / new data.

Every RTOS will have some I2C/SPI driver implementation.
In I2C case it would be great way to get rid of current (ugly) code.
SPI flash will benefit greatly from DMA, other performance gains will be probably quite small (or there will be small penalty)

blckmn, 2016-11-27:
@ledvinap check out bus_spi and i2c both have while loops waiting for response following request.

The issue is reads from these are blocking... rather than genuinely using the ISRs.

borisbstyle, 2016-11-27:
So what rtos would be the best choice?

Also on F1 boards we are running out of flash size.

ledvinap, 2016-11-27:
@blckmn : Yes, but bus is not released in this time - RTOS may preempt bus thread during this time (once for each bus read/write op). From point of view of driver nothing is changed(but current code must be moved into separate function that is executed in bus thread) , RTOS and non-blocking bus driver implementation will handle the rest ..

blckmn, 2016-11-27:
@ledvinap blocking on bus yes that remains - but at the moment it's effectively blocking CPU too. RTOS I'd expect the while loop to relinquish control of CPU and use ISR to signal re-entry required.

blckmn, 2016-11-27:
Even at 10mhz SPI, we are wasting 16+ CPU cycles on an F4 waiting for a byte. i2c is worse... much much worse.

ledvinap, 2016-11-27:
@blckmn : RTOS yields current thread in I2C function (read/write). When transaction is complete, ISR wakes the thread (makes it runnable), thread execution continues when no higher priority thread wants to run.
This is hidden inside I2C driver functions and transparent from point of calling code. There is no explicit yield/resume ...
(maybe we just don't understand each other ... )

martinbudden, 2016-11-27:
OK, I'm going to go against the grain here.

Why do we want an RTOS, other than because it gives us a nice warm fuzzy feeling?

When we talk about RTOS, we are talking about an RTOS kernel. They typically provide:
1. a priority based task scheduler, 
2. a minimalistic memory management subsystem, 
3. a set of timers, 
4. some task synchronization facilities ie semaphores/mutexes

Which bits of that do we need?

I would certainly say that if we went for an RTOS, we should avoid pre-emptive scheduling. That introduces all sorts of overhead. Cooperative scheduling, like we have, is actually very effective and underrated. This is especially true in our case where we have a single dominant task (the PID loop) and a bunch of other tasks that need to run infrequently (typically 100Hz or 10Hz).

Memory management. We don't need it - all memory is statically allocated and that is a huge benefit to us. No need to check for `malloc` failure or memory leaks. No need for garbage collection.

Timers. There is some benefit here.

Semaphores/mutexes - no need for them if we are doing cooperative scheduling. We already have atomic set/change variables and that is sufficient for our purposes (ie for clearing flags set by ISRs).

ledvinap, 2016-11-27:
@blckmn : Also speed is in bits per second, byte rate is eight time lower ... Task switch time is comparable, but full switch is necessary only after full transaction ...

blckmn, 2016-11-27:
Im referring to the inefficiencies that exist right now - going to RTOS will alleviate this as you say - one of the key benefits in my view. I'm pretty sure we are on the same page - I'm all for moving ASAP - and taking the approach you suggest.

blckmn, 2016-11-27:
Yes you're right so it's actually about 130cycles wasted at 10mhz spi. Which is why CPU use goes through the roof if you do nothing but lower spi speeds...

blckmn, 2016-11-27:
@martinbudden my vote for a move is purely background tasks - though that said rewriting the spi and i2c drivers to free up CPU can be done but it means moving mpu, baro code etc to not expect blocking calls.

ledvinap, 2016-11-27:
@martinbudden : 
Preemptive scheduling is quite fast - task selection is same as in cooperative case, task switching only needs to save/restore registers. ~1.3us is achievable for context switch, 600ns seems possible on F4.

Coding with cooperative scheduler is simpler, but mostly because you are ignoring dependencies. Blocking interrupts to sync with IRS code may be less efficient then proper preemptive mutex - mutex block other thread only when there is collision / all interrupts (of given priority) are blocked  

Also splitting realtime gyro task from rest of code will improve it's performance - it will decrease jitter, remove timing constrains on low-priority tasks(which can be cooperatively scheduled) etc.. 

GPS code will be greatly simplified if it is run in separate thread ..

martinbudden, 2016-11-27:
@ledvinap , when I speak of the overhead of pre-emptive scheduling, I'm not talking about the time taken for task switching, I'm talking about the overhead of managing access to shared data. Data is shared all over the place in betaflight. If we had pre-emptive scheduling we could have things like the PID task interrupting the RX task when, say, half of the RC channels had been updated.

martinbudden, 2016-11-27:
@blckmn , you say "so it's actually about 130cycles wasted at 10mhz spi.". I say: "so what?". CPU cycles are not a scarce resource for us, so it does not really matter if we waste a few.

Don't forget our goal is high performance, not high efficiency. Often high performance comes at the cost of reduced efficiency. Let's not forget our priorities here.

We waste lots of RAM as well. Take for example the `diff` command. That is hugely wasteful of RAM, but who cares? We have plenty of RAM, so use it to the full.

Talking about an RTOS is talking in solution space. We need to go back to problem space: what is the actual problem we are trying to solve?

blckmn, 2016-11-27:
Yes that's true @martinbudden - but we are pushing the F3 now harder and harder - so CPU will (you might say is - at higher gyro rates) become scarce. I'm all for not prematurely optimising. I'm actually happy to look at updating the spi i2c drivers to free up some CPU - even before rtos - if it helps.

digitalentity, 2016-11-28:
We will free up a lot of CPU if we do proper asynchronous background bus operations. I'm slowly moving towards making this happen. Hopefully, after having INAV 1.4 out of the door I'll look deeper.

@martinbudden RTOS will make things more organised in the long run, but the whole design of the system will have to change to make use of RTOS efficient, otherwise we will suffer a huge performance penalty of high tick rates and frequent context switches. Also, RTOS doesn't natively solve the issue of doing bus operations in the background - having a separate thread for bus driver only masks the problem without solving it. SPI wait for bus transfer takes 130 cycles per byte (~1.8us), context switches from and to the bus thread will take roughly that time so we are really saving nothing.

I like RTOS for the order it brings, but I don't see a real benefit in using RTOS in Betaflight/INAV.

martinbudden, 2016-11-28:
> but the whole design of the system will have to change to make use of RTOS efficient

Exactly. We can't just shoe-in an RTOS and expect to get automatic benefits.

> SPI wait for bus transfer takes 130 cycles per byte (~1.8us), context switches from and to the bus thread will take roughly that time so we are really saving nothing.

Agreed.

Don't get me wrong, I'm not against the idea of an RTOS in principle, I just disagree with the idea that using one will "magically" make betaflight better. Converting to an RTOS will involve a huge amount of work just to get to where we currently are.

digitalentity, 2016-11-28:
@martinbudden I believe we have had this discussion in Cleanflight repo about a year ago and came to same conclusion - RTOS in principle is good, but is not a Holy Grail that will magically make the firmware better.

IIRC, after that discussion our cooperative scheduler was born.

ledvinap, 2016-11-28:
> I'm talking about the overhead of managing access to shared data. Data is shared all over the place in betaflight. If we had pre-emptive scheduling we could have things like the PID task interrupting the RX task when, say, half of the RC channels had been updated.

There are quite a lot of possible methods to do this. For example double-buffering data into PID task should be quite easy. Or passing messages through queue(single-element queue is almost equivalent to double-buffering). Even using mutex (ideally with priority boost to prevent priority inversion) to protect shared data access should not be that hard to implement.

ledvinap, 2016-11-28:
Main advantage of RTOS in short-term is manageable bus arbitration/bus transaction in background + proven I2C bus driver. This will have quite small impact on current code base. And performance of I2C targets will be greatly improved.

In medium-term, gyro task separation may improve performance and simplify the code.
Splitting long interrupt handlers into task + small ISR will make new functions possible + improve performance (115200 softserial, maybe DMA-less DSHOT, better interrupt sharing between different functions)

bnn1044, 2016-12-02:
Keep in mind, Naked chip always run faster and more efficiency than the RTOS.
for long Run when we use higher clock MCU. this may not a problem. like people says, the code side will be more organized. But on the speed side, Without Rtos will be batter.

andre92, 2016-12-03:
agreed with bnn1044. I think more benefit can be had from doing I2C and SPI in the background without the CPU waiting for it, especially for F1 I2C boards.

andre92, 2016-12-03:
I2C/SPI in background sounds good. That way we can hold onto our F1 boards a little longer. Not sure if we would need RTOS for that, current task system seems to work fine. Maybe it is possible to make the scheduler pre-emptive without RTOS? Maybe that way we can run 4khz gyro sync on F1 I2C, and interrupt the attitude task if it doesn't finish on time.

digitalentity, 2016-12-03:
I have background bus operation in development. It's a low-priority project at the moment though.

ledvinap, 2016-12-03:
@bnn1044 : It depends .. There are cases when RTOS code is considerably simpler than equivalent non-rtos code. for example, in CF there are parts of code are implemented as state machine. On each call it is necessary to restore (and validate) state, dispatch some operation and store state. This is quite complicated and resource-hungry. With RTOS, state is implicit, it is saved and restored as simple push/pop of registers, which may be actually faster.

@andre92 : It is quite easy to write preemptive RTOS (It's interesting semester assignment). The hard part is removing all the bugs - that does take years .. Take a look at #1264/#1231 - `micros()` bugfix ... 

@digitalentity : IMO it will be a lot of work and resulting interface will be quite complicated.. My proposal above - thread per bus - is quite easy to implement and allows almost direct use of current driver code.

ledvinap, 2016-12-03:
BTW: can someone post performance measurements of current scheduler? What is per-dispatch overhead?

martinbudden, 2016-12-04:
@ledvinap , I'll try and post some figures within the next week or so. betaflight is nearing 3.1 and iNav is nearing 1.5 so there are some outstanding changes I want to make that take priority over doing measurements. There is some instrumentation in the scheduler now, so you could make some measurements yourself.

mkschreder, 2016-12-27:
The key here is the overall architecture. Even one thread per bus means nothing if the resulting design is bad from the start. The main problem that an rtos can solve quite well is the decoupling  timing of the io operations and the timing of all other logic. You can still continue to use the cooperative scheduler within the main fc code but use context switching when going between io code and purely algorithmic code. The takeaway is that io scheduling can be then optimized for best io performance while fc scheduling can be optimized for best fc performance.

One way to do io for example is to provide an api to a device that then talks to the io thread. An api function such as "write" may then dequeue a fixed size write buffer from a queue of available write buffers, fill it with data and then enqueue it into the queue of output requests for that particular device. When device driver processes the request it will enqueue the now free buffer into the queue of available buffers and so it goes. This means using only preallocated buffers and the size of these will regulate overall efficiency of io (bigger buffer will mean faster io but more memory used). This approach can be made lock free (it will still use appropriate synchronization primitives at queue level but the queue implementation is provided by the rtos so programmer does not need to worry about peculiarities of its implementation). Doing it any other way will require careful thought. I think using any kind of spinlocks should be generally avoided as much as possible because we have quite tight realtime constraints.  

PS. I think anyone thinking about partitioning up code into multiple concurrent tasks will benefit greatly from reading this excellent book about concurrency: http://kernel.org/pub/linux/kernel/people/paulmck/perfbook/perfbook.2016.07.31a.pdf

mkschreder, 2016-12-30:
Some issues I have seen so far while putting ninjaflight on top of FreeRTOS: 

- Global state is horrible. I have gone for a more encapsulated strategy with the only truly shared state being the config (I have also gotten rid of parameter groups completely and instead store config as a set of deltas in the flash - this ensures that config flash page erase happens very infrequently. Currently all objects get passed a pointer to a shared config. Not ideal but a lot better than parameter groups..). 
- Driver framework needs concurrency awareness. It is good to have blocking io (where you make an spi or i2c transaction and wait for result) but it has to happen while sleeping on the mutex so the moment a wait is initiated the rtos can reschedule to a higher priority task. For using an rtos, in the case of drivers, the code does need to employ proper locking (this needs to be done with care so as to never block an important thread for too long). 
- Gyro sync in particular can be simplified a lot by just adding a gyro_wait_for_update() function that sleeps on a semaphore that gets given to it from the gyro EXTI handler. This will just do a switch to the pid thread as often as gyro posts an update (1khz mpu "slow rate" or maybe even 8khz "fast rate" on a processor that can have enough processing power to do loop update that often.)
- There needs to be an ipc mechanism from the start. It needs to have topics to which an application can post an update. Each topic is a struct that can be read atomically by other threads and also contains a timestamp of last update. Every time a thread posts an update to a topic, other threads can then check if the topic has been updated since the last time they ran. Such an update is nonblocking and is treated as an "event". This is probably as good as it gets in terms of architectural design. Any kind of extended locking that may block a thread waiting for another thread should be avoided as much as possible when talking between concurrent modules.

So, while there is a lot more work to be done in cleaning things up for a truly concurrent firmware, at least I have reached the starting point of first baby step testing on top of freertos on an spracingf3 board..  I'll make a more detailed update when I feel happy with my code and have some hard measurement comparisons. The source is in my github repo.

martinbudden, 2016-12-30:
> I have also gotten rid of parameter groups completely 

Are you basing your work on Cleanflight or Betaflight? Betaflight doesn't have parameter groups yet - it's just got this horrible `masterConfig` structure.

I'm certainly interested in the work you are doing. I'll probably take time to look at your code once the betaflight 3.1 release is out.

borisbstyle, 2016-12-30:
@mkschreder Are you talking about this repo? https://github.com/mkschreder/ninjaflight

martinbudden, 2016-12-30:
@ledvinap - continuing the discussion...

Yes, I2C read is much slower. I'm getting timings of about 107us for gyro overclocked at 800kHz, so >200us for 400kHz gyro is about right.

I'm not totally against using RTOS. Rather my view is that making the changes will be quite a lot of work for no immediate "real" gains. From my own perspective there are other things I'd rather work on.

ledvinap, 2016-12-30:
@mkschreder : 
Can you elaborate on ParameterGroup problems?
Main goal of PGs was to make upgrading/config management easier, without affecting existing code (too much). The interface shall be quite generic, allowing different 'internal' implementations.

mkschreder, 2016-12-30:
Cleanflight. I hate parameter groups. :-) No global data please. I'll clean
up current branch, make sure it builds and post an update here soon so
everyone can play with it. I really don't like posting half done code but
that's the way it is right now.

@ledvinap I nuked parameter groups because in my opinion they hinder
options for managing and saving config. Also they introduce nasty
dependencies like needing a custom linker script making them more like a
pile of bricks dragging everything else down. Also I wanted to implement
delta compression for saving config in flash and current parameter group
code was just in the way. Additionally I also wanted to save config in an
emulated eeprom (a file) when running under sitl. Even that proved very
messy with current parameter group implementation. So I removed it.

Instead I have replaced it with a much simpler approach where all
parameters are defined together in a "config" module and a const pointer to
the config object is passed to objects that need to read config data. For
saving the config I use eeprom emulation similar to one described in an
stm32 app note here:
http://www.st.com/resource/en/application_note/CD00165693.pdf&ved=0ahUKEwjU2ObClJzRAhUTc1AKHWMWBSIQFggaMAA&usg=AFQjCNEs7ovKg6lDnBkdO9_H4pm-YVNmmg

So instead of rewriting the config page in flash (which wears flash because
of erase cycle) I start with an empty area and as user modifies parameters
such that they differ from previously saved ones I append these changes to
the flash as deltas. This works really well and only results in a page
erase after a really long time (when whole page is filled with deltas).

Again the idea is to make it both efficient and as simple to validate as
possible. I think my approach satisfies both goals.

We really need to make the code easy to unit test. So far I have managed to
refactor away almost all need for pointless "stubs" in unit tests. I must
admit I did break a lot of things in the process and some modules I have so
far simply commented out just to leave refactoring them for later. But then
again the repo boris mentioned is quite a drastic move in a totally
different direction (I don't want people flying it right now or using it
for anything serious because I really want to tighten all the bolts first).
So far I feel things are slowly coming back together again and prospects
for 100% validated code are pretty good with the groundwork I have been
doing so far.

Also another piece of news is that all the freertos code will be fully unit
testable. It builds natively and I have been evaluating prospects of
running natively compiled rtos code in the sitl build. It is slow due to
fully manual context switching but it works and means that rtos tasks can
be tested in sitl and then compiled without any modifications for the real
hardware.

ledvinap, 2016-12-31:
@mkschreder : 
> No global data please

PG allow to define config structure locally, you don't even need to publish it in .h file. Memory allocation is handled by PG, it can be easily changed without affection code using config structures.
Registration handles config saving, upgrading etc., but it treats PG as opaque memory.

>  they hinder options for managing and saving config.

It is list of memory chunks, each with tag, type and size. It is easy to iterate all PGs and handle them as you see fit. Only ordering is not strictly defined now, but that can be either fixed (sorting by PGID in linker) or sorted on access (with O(n^2) complexity)

> needing a custom linker script

Not necessarily, see implementation for darwin. Something similar is possible for all ELF targets (targets that support named sections), but using mechanism to append linker section seemed easier. 
 On real hardware linker script is needed anyway, so it is only added section.
This is price to pay for local registration. You can use array/structure in similar manner, but all registrations will have to be in single file. Or you may use different compiler extension (elf .init/.fini) to handle this.

> Instead I have replaced it with a much simpler approach where all parameters are defined together in a "config" module

That's the way CF used before PGs. Either all partial configd need to be declared in single file (separated from their module) or config.h depends on all modules - all configurable modules depend all others
Another problem is version upgrade - each new configurable parameter means that old config has to be discarded on upgrade. 

>So instead of rewriting the config page in flash (which wears flash because of erase cycle) 

STM devices are 10k erase cycles guaranteed, at extreme temperature, with 10 years retention. Real endurance at room temperature (developer conditions) is in 100k range (at extremely low temperatures this will still be in tens of thousands). Also FLASH failure will be handled on boot, so it is unlikely to destroy whole drone.
Delta compression is possible with PGs, just store PGID+offset (and store full record first to allow upgrades).

mkschreder, 2016-12-31:
I still think it is overengineered to an extreme. In my opinion parameter
restore after an upgrade is the job of the configurator. Configurator
should pull all params over msp, reflash and then restore. If there is any
conflict when doing that then user can see any errors and the restore can
be aborted..

One particular issue is ease of unit testing. With a config pointer passed
to an object you just define a config on the stack, reset it, set values
and pass it when creating the object to be unit tested. Then each test
knows what state it is starting with. Having some global state somewhere in
a linked in elf section is horrible for any kind of validation. All state
needs to be controllable and strictly defined for easier validation.

martinbudden, 2016-12-31:
@mkschreder comments on your comments:

> Also I wanted to implement delta compression for saving config in flash

I wouldn't consider this a priority. Config data is relatively small about 2K. And even if you do decide to do compression, I don't think PGs make it that much more difficult.

> Instead I have replaced it with a much simpler approach where all parameters are defined together in a "config" module and a const pointer to the config object is passed to objects that need to read config data.

As @ledvinap says, that is the approach Cleanflight used to use (and betaflight still does, although it is moving away from that). There are significant problems with this. In particular it breaks data encapsulation which is not good programming practice.

> We really need to make the code easy to unit test. So far I have managed to refactor away almost all need for pointless "stubs" in unit tests.

Agreed, the code needs to be easy to unit test. The current test system is far from optimal. And having to create all those stubs is annoying and slows testing down. But none of that is really related to parameter groups.

mkschreder, 2016-12-31:
It is a corner case for data encapsulation. It can be seen as the same type
of data dependency as you have between an application and a database. Both
have their own representation of the same data and database ultimately
models data structures that are specific to an application.

Parameter groups will only work if they are not global. In my sitl I can
have multiple quads running in the same application - ultimayely with
different configs. Achieving full data separation between them is not only
necessary for allowing them to coexist in the same application but also is
a feature of good design.

martinbudden, 2016-12-31:
> In my sitl I can have multiple quads running in the same application - ultimayely with different configs.

Yes, parameter groups were never designed to handle that situation. If you are simulating multiple quads then indeed having a single config object per quad is a perfectly valid design option. But I think it is not such a good option for the case of a dedicated flight controller for a single quadcopter.

mkschreder, 2016-12-31:
The thing is, it is still good design. Adding a constraint like multiquad
sitl only forces one to do things right (it is not strictly useful unless
one wants to experiment with ai navigation). When I started refactoring
cleanflight, a lot of things were already quite separated and could be
easily restructured into objects. The last and most notorious itch were
these parameter groups that are used everywhere and introduce a dependency
into a lot of modules on a state that is not instantiated like the object
itself. Suddenly PGs did not look like a good idea at all.

The thing is both sitl and unit tests serve as a validation platform and if
as many things as possible can be tested on them without any modifications
then it is only a good thing.

ledvinap, 2016-12-31:
> In my sitl I can have multiple quads running in the same application - ultimayely with different configs.

For most code, PG interface is function all that return pointer to requested structure (possibly indexed for _ARR). Either this function can take use some simulation-local state to check correct registry, or it may be possible to pass configID to all PG  access functions (that will be almost identical to your way of passing used config pointer).
With the exception of explicit configId parameter, all can be implemented without affecting code using PGs . Just change few macros, implement suport functions and you are done.

ledvinap, 2016-12-31:
As for unittests - with PG you get default configuration implicitly, which is probably an advantage.
Keeping state consistent may be a bit harder, but PG reset followed by setting of parameters test is interested in should do in most cases.
Also with PG you gent only config parameters from modules you link. With global config approach, test  is dependent on all modules.

martinbudden, 2016-12-31:
Also PGs could be extended to support multiple instances of a flight controller for use in a simulator. The semantics would be quite neat, you would just use, eg, `rxConfig(objectHandle)` rather than `rxConfig()`.

mkschreder, 2016-12-31:
No it is not dependent on all modules. In my tree I have moved all config
structs into config directory. Since config is like a database of sorts, it
is ok to make all modules depend on it (it consists of just headers).

Excluding some items in the config based on selected features can be done
of course. Unit tests just use a local "struct config" and depend on the
config headers (all the structs that define individual settings).

mkschreder, 2016-12-31:
That is called overengineering the obvious. I'm also quite curious how that
would be implemented in practice.

My current approach is to instantiate a config, reset it, load delta from
eeprom and pass pointer to it when initializing instances of other objects
in the system.

mkschreder, 2016-12-31:
One nice side effect about only saving a delta in the "eeprom" is that
technically one can do away with an eeprom area smaller than the size of
the full config. As long as one promises to only ever make changes to a
smaller percentage of parameters and not all of them. Not strictly useful
maybe but just pretty neat. A side effect..

ledvinap, 2016-12-31:
> That is called overengineering the obvious. I'm also quite curious how that
would be implemented in practice.

Implementing this is quite trivial, adding the parameter to identify config is only global change (or you can use thread-local tag to identify config and avoid the parameter).
Just change `PG_DECLARE` macro to define:
```
rxConfig_t* rxConfig(void* objectHandle) { return getConfig(objectHandle, PG_RX_CONFIG); }
```
PG initialization will scan PG registry, dynamically creating `PG_ID` -> `malloc(pg_size)` map (even C++ is not a problem here).

ledvinap, 2016-12-31:
Problem with your central config is that it is separated from modules using it. 
Another problem is how to reset it? Either reset functions has to belong to config (separated from code using it) or dependency on all modules remains.

mkschreder, 2016-12-31:
Dependency on the config schema yes, but not on the module code itself.
Defaults are part of the config schema. Just like you can have defaults as
part of a database schema. In the case where the same default config is not
valid for different board configurations, board init can reconfigure the
defaults on first boot by just altering the config and then save the config
to flash. By default the config is initialized to "the most reasonable" set
of defaults that should work unaltered for most configurations.

mkschreder, 2016-12-31:
Yes all reset functions are part of the config module. Not counting board
init that *may* alter defaults on first boot (currently I have not seen
where this may be necessary). Looking at the database analogy, all defaults
and reset functions are part of the schema. Not part of the application
that uses the database (flexibility comes from application being able to
alter invalid defaults. This requires explicit checks by the application
upon first boot).

mkschreder, 2016-12-31:
I'm making new year's dinner and thinking about how to best separate
drivers and fc logic into two distinct rtos threads and have fc logic run
during the lengthy i2c transactions and other io ops.. Happy new year :D

mkschreder, 2016-12-31:
It is a doable approach, considering that current sitl system and fc
already do run in two separate threads. The question is how much more
efficiency can be gained from doing it on the real hardware.. It definitely
needs to be tried..

mkschreder, 2017-01-05:
I have completed first dirty test with running pid loop as a high priority rtos task. 

Code: https://github.com/mkschreder/ninjaflight/blob/master/src/main/fastloop.c#L46-L154

Considering the requirement of running the whole pid loop as quickly as possible I have opted for a "fastloop" approach where the whole inner loop runs as a high priority rtos task. Using this approach the pid loop is clocked by the gyro update interrupt and can not be preempted by other tasks since it is the highest priority task. However it yields the cpu to other tasks during i2c ops and while waiting for next gyro update. To me this is as good as it gets. Results are clearly observable by msp being quite responsive even at the highest gyro update rate (I could not go 8k with i2c cause i2c operation takes longer than 125us, but 4k seems to run pretty smoothly). 

So I think using an rtos is very beneficial for meeting the tight timing constraints. Basically separating an application into a number of concurrent tasks ensures that time domains of the different tasks are fully separated. This is impossible with single loop approach. With an rtos one can ensure that high priority pid loop always gets the chance to run while not having to worry about the complexity of other tasks - they will always finish eventually but they may be arbitrarily split up over several iterations of the pid loop. 

My summary of the steps I did: 

1. separate imu + pid + mixer into a special high priority thread that is clocked by the gyro through the use of a semaphore that is given to it in the EXTI irq handler for the gyro. 

Whenever data is ready, the gyro will generate an interrupt. While there is no interrupt, the cpu does other things and the high priority "fastloop" thread is suspended so it never runs until semaphore is given to it. As soon as the interrupt occurs, the irq handler gives the semaphore and directly switches context to the high priority fastloop thread regardless of what cpu has been doing. For highest efficiency it is important that none of the peripherals that are in use by the fastloop, are used by any other threads because we want to have as few obstacles as possible for the fastloop so that it can complete in shortest possible time.

2. use i2c delay between starting i2c transaction and until it is done as another place to do other things (this improves performance considerably). This gives a 140us time window during the gyro read operation to do other things while also servicing the i2c interrupt. Unfortunately f3 i2c code does not use interrupts yet so I was not able to recover this dead time on f3, but my loop still ran fine because I used the time between completion of fastloop iteration and the next gyro interrupt to run other tasks. I also split up the fastloop such that when it reads accellerometer, it only stores the value and yields - waiting for next gyro interrupt and then "catches up" with the gyro. 

The way to recover dead time in i2c transactions so that it can be used for running other tasks is like this:

Instead of doing this:

    timeout = I2C_DEFAULT_TIMEOUT;
    while (busy && --timeout > 0) { ; }
    if (timeout == 0) {
        return i2cHandleHardwareFailure();
    }

You would do this:

    if(!xSemaphoreTake(_i2c_lock, I2C_DEFAULT_TIMEOUT)){
        return i2cHandleHardwareFailure();
    }

And in the irq handler you do this:

    BaseType_t wake = 0;        
    xSemaphoreGiveFromISR(_i2c_lock, &wake); 
    portYIELD_FROM_ISR(wake); // instruct rtos to do context switch after this irq is completed if there is a higher priority task waiting for this i2c transaction to complete.

Since the i2c transaction is initiated by the fastloop and since fastloop is the highest priority thread, the rtos will switch context to it as soon as the i2c interrupt returns. So the net effect is that our fastloop high priority thread is allowed to continue as quickly as possible *whenever it can*.

NOTE: in order to make this work I changed nvic priorities and adjusted freertos system call priority as well. I set i2c nvic priority from 0 (nonmaskable) to 1 and set maximum syscall priority to 1 as well. It is not possible to set it to 0 because that is not allowed for cases where you want to give the semaphore from isr (setting it to 1 gives it highest non-maskable priority). I also lowered the priority of various uart interrupts one step lower, such that i2c interrupt has highest priority among all of them. 

3. the code that belongs to the fastloop has to be self contained and isolated from all other code. This is easy to do when code is object oriented (struct based) and extremely messy to do when code uses global variables. Since I have already packaged up code into objects, this part was very easy to do. 

The main application communicates with the fastloop over a small memory area protected against concurrent access and the only operations that are allowed is to read and write the whole memory area as a single atomic operation. This avoids blocking the fastloop for any extended period of time. As always, the cost of concurrency is most often slightly higher memory usage because threads need to have their own copy of the data (locking would slow things down considerably).

The job of the fastloop is to basically do the low level motor control as quickly as possible. This can be implemented very efficiently using an rtos and I think it is the only clean way to do it when higher sample rates are possible. 

[... cut ...]
2. there is a memory overhead involved for storing multiple copies of some variables so that threads can operate unimpeded. I was able to fit it into the 20k on the naze but it is indeed pretty tight. I think the f3 with 40k ram and it's processing speed will benefit a lot more from an rtos. The only obstacle for me right now is absence of a proper interrupt driven i2c driver for the f3 which I either need to find somewhere or develop and test myself. Once i2c is interrupt driven, that will free up even more time for background tasks on f3 boards.

digitalentity, 2017-01-05:
@mkschreder great to have you looking into this, now we are getting somewhere! 

There is, however, one thing to consider - PID loop is not the one consumer of i2c bus time. Barometer, compass - they all require their share of bus time. How do you suggest we schedule this bus reads and do arbitration?

mkschreder, 2017-01-05:
The only problem as I see it is that fastloop will be delayed from running while i2c bus is busy. So highest update rate is limited by the sum of gyro i2c read time and the longest possible time that i2c bus can be blocked while doing something else (I think a baro read takes about 50us at most. So 140+50us + code time is the lowest possible looptime for this setup). 

Arbitration works like this:

Suppose fastloop is done with an iteration and tries to aquire the gyro sync semaphore. 

1. Gyro sync interrupt has not occured yet so semaphore is not given and rtos puts the fastloop task to sleep. It then switches to another lower priority task. Suppose it is baro task (or just the main application task that happens to read the compass+baro). Regardless, the task that now gets to run wants to read the baro. It starts an i2c transaction and grabs the i2c lock (we actually would have two locks: one mutex for the i2c bus as a whole, and one semaphore that is given from i2c interrupt to signal completed transaction). The i2c main lock is free so that thread grabs the lock and proceeds. 

2. An i2c transaction on behalf of the lower priority thread is started. I2C bus then handles several interrupts involved in the transaction while rtos either runs another application task or runs the rtos idle task. i2c interrupts preempt whatever task is running but do not result in a context switch. A context switch *may* occur if rtos timer expires and rtos determines that there is another task that needs to run. Otherwise whatever is running continues to run. The baro reading task now sleeps on the semaphore that will be given to it once i2c interrupt signals completion of the i2c transaction. 

3. While all this is happening, a gyro sync interrupt arrives and rtos determines that fastloop needs to run now. It does a context switch from whatever else has been running and switches context to the fastloop task. Fastloop then starts with trying to read the gyro. It tries to start an i2c transaction but since the i2c main mutex is locked, rtos puts it back to sleep again and continues whatever it has been doing.

4. i2c interrupt determines that i2c baro transaction has completed and i2c interrupt signals the "done" semaphore on which the baro task has been sleeping. RTOS directly unblocks the baro task (because of priority inheritance in freertos, baro task now has highest priority because fastloop is sleeping on the same i2c mutex that the baro task is holding as well). Baro task continues in the i2c read call and unblocks the i2c mutex. RTOS now resets the priority of the baro task back to normal priority and since the fastloop is now runable and has highest priority, another context switch is done to switch back to the fastloop. Fastloop now being the highest priority task, continues to run regardless of any other events, claims the i2c mutex and starts the gyro read transaction before being switched out again. 

A correct arbitration would go something like this: (I need to double check once I reenable both compass and baro and see what looptime I get when they contend for i2c bus together with fastloop task). 

1. gyro sync arrives, context switch to fastloop
2. i2c mutex held, context switch to something else
3. i2c completed, context switch to baro
4. i2c mutex release, context switch to fastloop
5. fastloop starts i2c, switch to something else

best case:

1. gyro sync arrives, context switch to fastloop
2. fastloop starts i2c read, context switch to something else

Obviously to get best performance it is necessary to have compass and baro hooked up to a separate i2c bus from gyro. In such case, both baro/compass and gyro read can happen simultaneously and without any waits. But then it is probably better to just use the boards that have spi acc/gyro if 8k updates are required... 

Another approach is to read the compass as part of fastloop. Baro does not really belong there, but compass is directly used to update the dcm so it probably would make sense to read it as part of the same loop. Then no arbitration is necessary - at least not between gyro+acc+compass. Then compass can be read as part of the same tick skipping scheme that I'm testing right now for interlacing acc reads at high rates. Basically when compass is read then gyro read and subsequent dcm update is skipped and the time *dt* for next gyro update is doubled to account for skipped update. 

Mutexes: http://www.freertos.org/Real-time-embedded-RTOS-mutexes.html
Semaphores: http://www.freertos.org/Embedded-RTOS-Binary-Semaphores.html

[3 more replies]