betaflight/betaflight #1682: all empty resource from NONE to A00???

TheAngularity, 2016-11-27:
@mikeller @martinbudden 
the latest #1622 fills empty resources from NONE to A00 ... I think that should not be so

before merge:
```
# Betaflight/SPRACINGF3EVO 3.1.0 Nov 27 2016 / 13:29:45 (417d27d)

# name
name -

# resources
resource BEEPER C15
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 A06
resource MOTOR 6 A07
resource MOTOR 7 B00
resource MOTOR 8 B01
resource MOTOR 9 B10
resource MOTOR 10 B11
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM A15
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP A08

# mixer
```

after merge:
```
# Betaflight/SPRACINGF3EVO 3.1.0 Nov 27 2016 / 15:06:06 (18b052e)

# name
name -

# resources
resource BEEPER C15
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 A06
resource MOTOR 6 A07
resource MOTOR 7 B00
resource MOTOR 8 B01
resource MOTOR 9 B10
resource MOTOR 10 B11
resource MOTOR 11 A00
resource MOTOR 12 A00
resource SERVO 1 A00
resource SERVO 2 A00
resource SERVO 3 A00
resource SERVO 4 A00
resource SERVO 5 A00
resource SERVO 6 A00
resource SERVO 7 A00
resource SERVO 8 A00
resource PPM A15
resource PWM 1 A00
resource PWM 2 A00
resource PWM 3 A00
resource PWM 4 A00
resource PWM 5 A00
resource PWM 6 A00
resource PWM 7 A00
resource PWM 8 A00
resource LED_STRIP A08

# mixer
```

mikeller, 2016-11-27:
It's weird that this changeset would affect the resource allocation. But it could have been caused by the config becoming stale. Have you tried resetting the config to defaults to see if it still shows unassigned functions as assigned to A00?

TheAngularity, 2016-11-28:
both config are resetting to defaults and that the output after this

blckmn, 2016-11-28:
It looks like there's no check in printResource for if the iotag_t is valid...

ledvinap, 2016-11-28:
Also `DEFIO_TAG_ISEMPTY` is internal abstraction, IO is designed so that both `ioTag_t` and `IO_t` evaluate to false when empty - `if(io)` / `if(tag)` is correct way to test for non-empty tag/io

blckmn, 2016-11-28:
https://github.com/betaflight/betaflight/pull/1693 just added this but won't be able to test on hardware until tomorrow.

TheAngularity, 2016-11-28:
merged your latest #1693 ... flashed ... after resetting back to defaults the unassigned functions shows NONE back again ... thx @blckmn

closed 2016-11-28 by TheAngularity