betaflight/betaflight #1685: Changing values with MSP over SmartPort are not stored into EEPROM

basdelfos, 2016-11-27:
When using the LUA script on a Taranis to change PID values, they seem not being stored into EEPROM. Changing PID values works, but after reconnecting the lipo, the changed values are restored to original value before changing it with the LUA script.

Steps to reproduce:
1. Install BF 3.1 (build of master of date 2016-11-27) and BFSetup.lua on a Taranis.
2. Start LUA script on the Taranis.
3. Change a PID value.
4. Save changes by pressing Menu button for 2 seconds, select Save.
5. Reload settings to see if values are stored in BF.
6. Disconnect and reconnect lipo.
7. Start LUA script again
8. PID values are restored to initial setting (as before step 3).

Gear used:
- Taranis with OpenTX 2.2.0 RC7
- X4R
- KISS FC @ Betaflight 3.1 build (master from 2016-11-27)

basdelfos, 2016-11-27:
Confirmed this also on the workbench with BFC. Values are saved while lipo or USB is connected. After disconnecting/reconnecting lipo and/or USB, the values are reset to initial value. Same applies to the rates.

jflyper, 2016-11-29:
> Confirmed this also on the workbench with BFC. Values are saved while lipo or USB is connected. After disconnecting/reconnecting lipo and/or USB, the values are reset to initial value. Same applies to the rates.

I can't reproduce this.

basdelfos, 2016-11-30:
Did you try to reproduce it on a KISS FC too? Could be target specific.

jflyper, 2016-11-30:
No KISS FC here.

philippamlong, 2016-12-02:
I repruduced it on SPF3 with latest build. Storing EEPROM using stick command (left: down left; right: down right) works after changing the settings in lua screen.

basdelfos, 2016-12-02:
Damn, i feel stupid, didn't know i have to store the changes with the stick commands. Now it works indeed. Maybe out of user perspective it is better to change 'Save settings' to something like 'Submit settings' in the LUA script.

Issue can be closed.

philippamlong, 2016-12-02:
Maybe we should not close this one yet. I just tried the stick command as a workaround. I think, the lua script should save instantly, it would be better for usability.

raphaelcoeffic, 2016-12-25:
Are you sure the script used was including the changes commited on Nov 15 (don't know when it was merged)? These changes should save the changes into eeprom each time (save command is sent).

basdelfos, 2016-12-25:
Just checked and I indeed had an older version than Nov 15 (had Nov 11). Double checked if I had the issue still and it is still there. Than installed the newest LUA script from Github and the issue is gone. Saving settings from the Taranis now also stores into EEPROM.

Thanks. Don't have to remember to store with stick commands after changing PIDS/Rates via S.Port.

I will close the issue as resolved.

closed 2016-12-25 by basdelfos