betaflight/betaflight #1689: 3.1 Nov 27th build... YIKES

QuadroBro, 2016-11-27:
This build is a disaster from a PID's perspective. Nothing but high freq oscillations until I tuned P's to low 30's I's to low 20's and D's to 10/12. Even then, there was crazy propwash on hard turns. I had to give up on tuning when I realized that D setpoint weight was having no effect whatsoever.

Additionally, has anyone else noticed that the ability to tune via the OSD disappeared??? I'd be more inclined to try to tune if I could use the OSD, but... 

Running Dshot 600, Tornado T2 motors (requiring down-tuning due to power and efficiency but this is over-the-top) and omnibus F4

borisbstyle, 2016-11-27:
D setpoint is not meant to tune oscillations with it, but stick feel.

Did you erase eeprom?

QuadroBro, 2016-11-28:
@borisbstyle It said erasing, so I'm assuming so. If it's not setpoint, then I'm not sure what it is as it feels like the pid's have been amplified considerably. Again, this is on 3 quads (2x omnibus F4 and 1x omnibus F3 Pro). I saw another thread stating the dude blew a motor today and, granted, he was running the wrong protocol, but I'm pretty sure if I ran at stock pids for more than a few minutes, I'd be in the same boat.

sasofpv, 2016-11-28:
I experienced same issue today with build #759 with Soul F4 and Aikons 20A (16.5) with DSHOT300.

Crazy high oscilation. I reflashed board and left default PIDs with same results. I was previously running build #737 without any issues.

martinbudden, 2016-11-28:
@QuadroBro , could you give a link to the build you used that has the problems?

And if you have time, it would be great if you could track it down to an exact build, ie something like build 753 works, but build 754 doesn't work. This would greatly help in fixing the problem.

sasofpv, 2016-11-28:
I will try to find exact build  when weather gets better...

martinbudden, 2016-11-28:
That would be great.

MJ666, 2016-11-28:
I was flying an build from Saturday's master on 4 different AlienFlight F3/F4 based controllers without seeing any issues. Two Quad and two Hexacopters.

benwilliamson86, 2016-11-29:
I am having same issue. Flashed approx 12 hours ago. Its like someone cranked the P waay to high even though its on defaults. Running Dshot on 2206 Tornado T2's.

sasofpv, 2016-11-29:
I tested again with build #763 (quick hover test in garage) and problem seems to be gone. I need to tested it again but looks OK (I experienced oscillation on build #759 during hover test too).

Does it have anything to do with PT1 filter initialization?

martinbudden, 2016-11-29:
Yes, the failure in the PT1 initialisation could have caused exactly these symptoms.

@QuadroBro , can you retest with the latest build and confirm the problem is fixed?

benwilliamson86, 2016-11-29:
Fixed for me. Many thanks.

sasofpv, 2016-11-29:
I flew a couple of lipos today with #759 and it works great. No issues at all. Even blackbox logging and transfering log to computer works now. :)

Thank you!

QuadroBro, 2016-11-30:
Fixed for me as well using the build from this morning. I pasted my old pids back in and it flies great!!!

borisbstyle, 2016-11-30:
You cant see the build number when connecting to the board.

Issue closed as it seems resolved

closed 2016-11-30 by borisbstyle

martinbudden, 2016-11-30:
Thanks for testing everyone.

@QuadroBro , to get the build number, type `version` in the CLI, you'll get something like:
```
# version
# BetaFlight/OMNIBUSF4 3.0.1 Oct 18 2016 / 10:22:03 (48b7b4f)
```

That last number in brackets is what is needed to identify the git submission.