betaflight/betaflight #1695: Safer arming mechanism to prevent accidental arming

rodgerramjet, 2016-11-28:
Unfortunately i have a spektrum so cant setup 2 switch arming like you can using taranis's mixer- as far as i can tell.

There doesnt seem to be a way to set this sort of condition in the bf configurator's modes tab- you can add multiple 'or' conditions but not 'and' conditions. Is it possible to get this?  

Or if its simpler - an option something like having to arm twice (within 2 seconds) after you first plugin battery? (Could just be cli variable 'double_arm' set to false by default)

mikeller, 2016-11-28:
What is the reason you cannot use switch arming with your Tx? (Sorry, I don't know spektrum.)

One potential option for making stick arming more secure would be to require the sticks to be held in the 'arm' position for a certain amount of time (2 seconds). Your thoughts?

rodgerramjet, 2016-11-28:
@mikeller 

Just for clarity- i can use switch arming but no option to make it a '2 switch' process for safety (eg bump tx accidentally)

Ive searched the net on how to setup a 2 switch arm with my dx8 and cant find any info. 

Taranis has plenty of guides on how to do it- it allows you to setup logical switches which i dont think spektrum does. Heres a bardwell vid on taranis setup https://youtu.be/bv3VJ1jznw8

Is the option of pressing the same arm switch twice (on-off-on) within 2 secs not possible or a bad idea?(just for first arm)

flounderscore, 2016-11-28:
Depending on what Spektrum TX you have, it is possible. Create a mix for your arming channel (mix it with itself) and use your secondary arming switch to enable the mix.

That approach works fine for me on a DX7s (although I use it to have two disarm switches, flight mode and throttle hold - trying to mimic my heli setup).

mikeller, 2016-11-28:
@rodgerramjet As an alternative to @flounderscore's suggestion (and not knowing if the spektrum firmware supports this), you could also create an additive mix of 2 switches and scale them down to 50%. This would then effectively be transmitted as a '4 position switch' on the channel, allowing you to assign the 'both up' range to arming.

rodgerramjet, 2016-11-29:
@flounderscore @mikeller 

wow thanks for your help- you guided me in working this out using mixing on DX8... i didn't realise using the same channel was the way

I set the settings to GER->GER (my arming switch).. rate -125%.... offset -95%... Sw Mix1 (the other switch)

Cheers.... I'll close this issue.

closed 2016-11-29 by rodgerramjet

macht-nix, 2017-04-19:
Despite knowing all that... & #458, today i managed to cut myself..... (it´s not too bad, i´m ok)
With pain comes ideas:
I think the easiest and best solution for the problem would be to make "Safer arming" accessible and TX independent by hitting just one checkbox in the betaflight configurator to set it up.
And it´s already nearly there!
Just combine switch and stick arming.

**arming prozedere would be:**
1. Arm the copter with switch
2. Motors do not spin (maybe buzzer is beeping)
3. Do the stick arming
4. Motors start to spin
5. Disarm the copter with switch

mikeller, 2017-05-22:
@macht-nix: In a way your proposal combines the negative points of both stick arming and switch arming:
1. it requires a channel for the switch, and set up for the channel;
2. it requires stick input at zero throttle to be disabled.

If we require 1., then we can as well ask the users to set up a channel that switches value only after 2 switches are actuated. Or it's probably worthwile making a feature request to OpenTx / er9x to add a 'template function' for an 'arming' function that, when assigned to a channel, requires 2 switch actuations.

rnelias, 2017-05-22:
the simplest way to make the arm process safer is put your throttle stick at 100% when FC is disarmed. 

BF will only arm if throttle is at 0% and (if you choose so) use a specific switch...

mikeller, 2017-05-22:
@rnelias: With most of the common transmitters, you can one-up this in 2 minutes, without having to use anything that is not already in there: Just define your arming channel to be the latching function of a momentary switch that is enabled by a two state switch. This way, in order to arm, you have to switch a switch and then pull the momentary switch.

I think the problem is more about people not willing to spend time on getting it set up properly, than about lack of possibilities to set it up safely.

macht-nix, 2017-05-29:
@mikeller: if i had a radio that allowed me to use a momentary switch and make it a sticky switch i would use it but my DX9 is just able to mix regular switches… (i mix 3 switches to arm my copter but if i don't switch them back after disarming its all useless)..… and there are a lot of other radios out there that are not opentx ….

...if you would ask for 2 switches to be actuated it would require 2 extra channels for arming (on some radios one to much) and some radios are not able to set a momentary switch to a channel.
thats why my idea was to kind of replace the sticky switch with the stick arming gesture and for the disarming just switch back the arming switch

rodgerramjet, 2017-08-01:
looks like BF 3.2 added this as a mode :)