betaflight/betaflight #1718: Massive increase in file size

TheAngularity, 2016-12-01:
using targed SPRACINGF3EVO ... merged latest commits from 2f0f23e to 8af1508 
now the file size has increased by 220KB ... before file size is 471KB after merging it have 691KB
i think there is something wrong :((

Wind0r, 2016-12-01:
The filesize increase is on purpose.
Betaflight now do loop unrolling and other stuff. This grows the binary but reduce cpu time.

TheAngularity, 2016-12-01:
Wow, ok ... there are the 256KB flash almost full 
Further I've seen the gps and mag were removed from the evo ... is that right

martinbudden, 2016-12-01:
Yes, ROM is now becoming an issue on F3 targets.

Closing this.

closed 2016-12-01 by martinbudden

Wind0r, 2016-12-01:
GPS is removed from all F3 targets and Mag is also removed from the evo, but this stuff is work in progess. 
I think within the next weeks/months binary size will shrink again and those features maybe come back.

TheAngularity, 2016-12-01:
I do not miss it ... from me we can also removed the baro ;)
also a very impressed reduced cpu time ... now i can use the highest pid loop with an cpu load from 37%

SandroGrzicic, 2016-12-04:
Just did a comparison of the HEX size, with the latest master it's 690KB, with MAG enabled it's 697KB, maybe I'm doing something wrong here? 7KB doesn't seem like a huge number, but maybe it is? What's the actual limit?

I of course understand that MAG/GPS/BARO/etc. features are not the priority in BF, especially since 3.1 is not actually out yet.