betaflight/betaflight #1728: blheli configurator bricked esc's flashing thru spracingf3/Bf 3.1.0

beachbreak, 2016-12-02:
flashed my foxeer f303 with BF 3.1.0 775
went to flash my xs 30A (no caps) with Bheli using configurator...both previously worked with earlier dshot FW....and during the flash, esc stopped loading, and now unable to reflash...bricked.  
tried with a different esc/same FC, same deal, stopped flashing quarter way through.

DieHertz, 2016-12-02:
Why you're reporting it to BetaFlight repository?
Have you used "Save Log" button to save logfile when you ran into issues?

Norcal3d, 2016-12-05:
This seems to be an issue on some SPRacingF3 targets. You cant flash ESC via passthru on BF3.1. Flash back to BF3.0.1 and you can flash the ESC just fine. Setting changes to hte ESC work fine in 3.1, just Flashing firmware is Broken.

DieHertz, 2016-12-05:
Would be nice to know what's exactly broken. My XRacer F303 v2 couldn't reproduce this issue while flashing XS30A. I've tried it multiple times, breaking signal connection occasionally, all attempts resulted either in a successful flash or **FLASH*FAILED** marked ESC which is easily fixed by a subsequent flash.

closed 2016-12-19 by borisbstyle